raulmur/ORB_SLAM2 #367: Wrong scale in the position output using stereo camera

gauxonz, 2017-06-29:
## Problem description
I test my own stereo cam with the `stereo_euroc` example, and get the result trajectory with transition in wrong scale. The expected trajectory is around 30 meters but the transition in the output is always under 1 meter. The trajectory is
[image] 
(the unit of both axis is meter).

## My setup and configuration
I calibrate my stereo camera using the [Caltech Calibration toolbox for Matlab](http://www.vision.caltech.edu/bouguetj/calib_doc/), and got the [.yaml config file](https://github.com/raulmur/ORB_SLAM/files/1112521/Drone03.yaml.txt). 

## What I have done try to solve and explore the problem and the test result
I hold the camera 1 meter above ground, looking straight down to the ground like this:
[image]

Then I modified the code to out put the initial Keyframe's Key Points and Map Points. The results shows that the depth of the Key Points has the mean of 2.8 meters, as long as the Map Points, which is obviously wrong. Here is some test output:
- Rectified stereo image with Keypoints
[image]

- Initial Mappoints
[image]
[image]
===================
So can anyone tell me where the problem is? Thanks a lot!!

YingjieLu, 2019-02-20:
Did you set the rectification as "true" in command line? I solved similar issues by doing that.

BristolVisualPFT, 2019-11-10:
@gauxonz I have the same issue as you using the stereo ORB-SLAM. Could you figure out what the issue was? Thanks

gauxonz, 2019-11-12:
@BristolVisualPFT It's mainly caused by the calibration. Try to do the calitbration with OpenCV's implementation and that solve my problem. You can check [my implementation](https://github.com/NLS-SJTU/stereo-calibration) on this is. Hope that can help.