raulmur/ORB_SLAM2 #371: XML error with openCV while launching mono ROS node

keshavchintamani, 2017-07-03:
Hi All,

I think some others have had this issue. I'm running Ubuntu 16.04 with ROS Kinetic. I think OpenCV 3 is somewhere on my machine. When I try to run the ROS node, I'm getting the following exception:

> ORB_SLAM2>rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt ../trinibot_core/settings/trinbot_camera_calib.xml /camera/image_raw:=/usb_cam/image_raw
> 
> ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
> This program comes with ABSOLUTELY NO WARRANTY;
> This is free software, and you are welcome to redistribute it
> under certain conditions. See LICENSE.txt.

> Input sensor was set to: Monocular
> OpenCV Error: Parsing error (../trinibot_core/settings/trinbot_camera_calib.xml(4): <opencv_storage> tag is missing) in icvXMLParse, file /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/core/src/persistence.cpp, line 2249
> terminate called after throwing an instance of 'cv::Exception'
>   what():  /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/core/src/persistence.cpp:2249: error: (-212) ../trinibot_core/settings/trinbot_camera_calib.xml(4): <opencv_storage> tag is missing in function icvXMLParse
> 
> Aborted (core dumped)

my xml file looks like this (I included a fake opencv_storage tag but that makes no diff with or without. 

  1 <?xml version="1.0" encoding="ASCII"?>
  2 <opencv_storage>
  3 </opencv_storage>
  4 <Camera_Matrix type_id="opencv-matrix">
  5 <rows>3</rows>
  6 <cols>3</cols>
  7 <dt>d</dt>
  8 <data>
  9 316.082876177544 0.0 152.389620768185 0.0 315.961717263906 118.531765605151 0.0 0.0 1.0
 10 </data>
 11 
 12 </Camera_Matrix>
 13 <Distortion_Coefficients type_id="opencv-matrix">
 14   <rows>5</rows>
 15   <cols>1</cols>
 16   <dt>d</dt>
 17   <data>
 18 0.157255 -0.292145 0.008454 -0.002377 0.000000
 19   </data>
 20 </Distortion_Coefficients>
 21 <opencv_storage>
 22 </opencv_storage>

The image_raw topic is streaming on usb_cam/image_raw. I verified this on the image_view.

Any info on what could be wrong?

poine, 2017-07-03:
Yes. You have two versions of opencv linked. Try 'ldd' on your program to
confirm it. Then make sure opencv 3 is found before opencv 2 in your
Cmakelists.
HTH

Poine

keshavchintamani, 2017-07-04:
Hi Poine, The ldd trace is not showing any links to OpenCV2 or any other version besides OpenCV 3.1. 

> ORB_SLAM2>ldd Mono
> 	linux-vdso.so.1 =>  (0x00007ffdf0b7c000)
> 	libroscpp.so => /opt/ros/kinetic/lib/libroscpp.so (0x00007f3fb6a50000)
> 	libcv_bridge.so => /opt/ros/kinetic/lib/libcv_bridge.so (0x00007f3fb682e000)
> 	libopencv_core3.so.3.1 => /opt/ros/kinetic/lib/libopencv_core3.so.3.1 (0x00007f3fb5ba2000)
> 	librosconsole.so => /opt/ros/kinetic/lib/librosconsole.so (0x00007f3fb596f000)
> 	libroscpp_serialization.so => /opt/ros/kinetic/lib/libroscpp_serialization.so (0x00007f3fb576c000)
> 	libpangolin.so => /home/kc/sandbox/3rdParty_builds/Pangolin/build/src/libpangolin.so (0x00007f3fb53e1000)
> 	libORB_SLAM2.so => /home/kc/sandbox/external-rospacks/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../lib/libORB_SLAM2.so (0x00007f3fb50a6000)
> 	libGL.so.1 => /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1 (0x00007f3fb4df0000)
> 	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f3fb4a6e000)
> 	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3fb4857000)
> 	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3fb448e000)
> 	librostime.so => /opt/ros/kinetic/lib/librostime.so (0x00007f3fb4262000)
> 	libxmlrpcpp.so => /opt/ros/kinetic/lib/libxmlrpcpp.so (0x00007f3fb4042000)
> 	libcpp_common.so => /opt/ros/kinetic/lib/libcpp_common.so (0x00007f3fb3e3a000)
> 	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f3fb3c36000)
> 	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007f3fb3a0f000)
> 	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3fb37f2000)
> 	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007f3fb35da000)
> 	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3fb32d0000)
> 	libboost_regex.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 (0x00007f3fb2fc8000)
> 	libopencv_imgcodecs3.so.3.1 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1 (0x00007f3fb2ba7000)
> 	libopencv_imgproc3.so.3.1 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.1 (0x00007f3fb11e7000)
> 	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f3fb0fcd000)
> 	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f3fb0dc9000)
> 	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f3fb0bc0000)
> 	librosconsole_log4cxx.so => /opt/ros/kinetic/lib/librosconsole_log4cxx.so (0x00007f3fb09a7000)
> 	librosconsole_backend_interface.so => /opt/ros/kinetic/lib/librosconsole_backend_interface.so (0x00007f3fb07a5000)
> 	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007f3fb03cd000)
> 	libGLEW.so.1.13 => /usr/lib/x86_64-linux-gnu/libGLEW.so.1.13 (0x00007f3fb014a000)
> 	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f3fafe10000)
> 	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f3faf881000)
> 	/lib64/ld-linux-x86-64.so.2 (0x000055bbd7e15000)
> 	libDBoW2.so => /home/kc/sandbox/external-rospacks/src/ORB_SLAM2/Thirdparty/DBoW2/lib/libDBoW2.so (0x00007f3faf672000)
> 	libg2o.so => /home/kc/sandbox/external-rospacks/src/ORB_SLAM2/Thirdparty/g2o/lib/libg2o.so (0x00007f3faf3db000)
> 	libopencv_calib3d3.so.3.1 => /opt/ros/kinetic/lib/libopencv_calib3d3.so.3.1 (0x00007f3faf07d000)
> 	libopencv_features2d3.so.3.1 => /opt/ros/kinetic/lib/libopencv_features2d3.so.3.1 (0x00007f3faedbc000)
> 	libopencv_highgui3.so.3.1 => /opt/ros/kinetic/lib/libopencv_highgui3.so.3.1 (0x00007f3faeb80000)
> 	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f3fae957000)
> 	libxcb-dri3.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 (0x00007f3fae754000)
> 	libxcb-present.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-present.so.0 (0x00007f3fae550000)
> 	libxcb-sync.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-sync.so.1 (0x00007f3fae349000)
> 	libxshmfence.so.1 => /usr/lib/x86_64-linux-gnu/libxshmfence.so.1 (0x00007f3fae146000)
> 	libglapi.so.0 => /usr/lib/x86_64-linux-gnu/libglapi.so.0 (0x00007f3fadf16000)
> 	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f3fadd04000)
> 	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f3fadb01000)
> 	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f3fad8fa000)
> 	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f3fad6f8000)
> 	libxcb-glx.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-glx.so.0 (0x00007f3fad4df000)
> 	libxcb-dri2.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-dri2.so.0 (0x00007f3fad2d9000)
> 	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f3fad0b7000)
> 	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f3faceb1000)
> 	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f3facca1000)
> 	libconsole_bridge.so.0.2 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.2 (0x00007f3faca9c000)
> 	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007f3fac639000)
> 	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007f3fac2a5000)
> 	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f3fac04c000)
> 	libwebp.so.5 => /usr/lib/x86_64-linux-gnu/libwebp.so.5 (0x00007f3fabdef000)
> 	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f3fabbca000)
> 	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f3fab956000)
> 	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007f3fab700000)
[... cut ...]
> 	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f3fa70da000)
> 	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f3fa6e7c000)
> 	libpcre16.so.3 => /usr/lib/x86_64-linux-gnu/libpcre16.so.3 (0x00007f3fa6c15000)
> 	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f3fa6a0d000)
> 	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f3fa679c000)
> 	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f3fa64f2000)
> 	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f3fa62cd000

keshavchintamani, 2017-07-04:
Hi all, apologies, clearly I took a wrong step with the settings file ! I misunderstood the settings file for the camera calibration xml in the opencv schema. Actually it needs to look like the Yaml file in the /Examples/Monocular/TUM1.yaml like so:

The only strange thing is the exception thrown was really misleading... so could be fixed for future users!

> %YAML:1.0
> 
> #--------------------------------------------------------------------------------------------
> # Camera Parameters. Adjust them!
> #--------------------------------------------------------------------------------------------
> 
> # Camera calibration and distortion parameters (OpenCV) 
> Camera.fx: 517.306408
> Camera.fy: 516.469215
> Camera.cx: 318.643040
> Camera.cy: 255.313989
> 
> Camera.k1: 0.262383
> Camera.k2: -0.953104
> Camera.p1: -0.005358
> Camera.p2: 0.002628
> Camera.k3: 1.163314
> 
> # Camera frames per second 
> Camera.fps: 30.0
> 
> # Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
> Camera.RGB: 1
> 
> #--------------------------------------------------------------------------------------------
> # ORB Parameters
> #--------------------------------------------------------------------------------------------
> 
> # ORB Extractor: Number of features per image
> ORBextractor.nFeatures: 1000
> 
> # ORB Extractor: Scale factor between levels in the scale pyramid 	
> ORBextractor.scaleFactor: 1.2
> 
> # ORB Extractor: Number of levels in the scale pyramid	
> ORBextractor.nLevels: 8
> 
> # ORB Extractor: Fast threshold
> # Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
> # Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
> # You can lower these values if your images have low contrast			
> ORBextractor.iniThFAST: 20
> ORBextractor.minThFAST: 7
> 
> #--------------------------------------------------------------------------------------------
> # Viewer Parameters
> #--------------------------------------------------------------------------------------------
> Viewer.KeyFrameSize: 0.05
> Viewer.KeyFrameLineWidth: 1
> Viewer.GraphLineWidth: 0.9
> Viewer.PointSize:2
> Viewer.CameraSize: 0.08
> Viewer.CameraLineWidth: 3
> Viewer.ViewpointX: 0
> Viewer.ViewpointY: -0.7
> Viewer.ViewpointZ: -1.8
> Viewer.ViewpointF: 500
>

closed 2017-07-04 by keshavchintamani