raulmur/ORB_SLAM2 #373: Performance with monocular tracking

chetanskumar, 2017-07-04:
I've been testing the ORB-SLAM framework with a several test sets. I noticed a pattern amongst the tracking of the keyframes: it often takes about 150 keyframes for the tracker to initialise and begin smooth operation. I decided to check if this was the case with the TUM freiburg1 desk set as well, and it so happens that the tracking begins approximately 150 frames from the beginning in this set as well.

I'd be grateful for help on whether this is expected behaviour for ORB_SLAM, and if not, how to go about fixing it.

AlejandroSilvestri, 2017-07-08:
@chetanskumar ,

Are you talking about the initialization process, before tracking and mapping starts?
Initialization highly depends on scene, and provides a coarse set of initial map points.  The first keyframes trigger bundle adjustments that can do that effect, changing many points abruptly.  This also depends highly on scene and correct camera calibration.

chetanskumar, 2017-07-17:
@AlejandroSilvestri,

I think that explains it. It's probably the Bundle Adjustments causing this effect. On another note entirely, what do you think of Dense Tracking and Mapping? From a practical perspective, do you believe its performance to be comparable to or better than ORB-SLAM?

AlejandroSilvestri, 2017-07-17:
@chetanskumar 

I'm sorry, I didn't try DTAM.  ORB-SLAM2 is specially interest because it registers features that can be used for other purposes than tracking, like relocalization, loop closure and new ones, and opens a field for new applications.

I believe there are many benchmarks on monocular slam.