raulmur/ORB_SLAM2 #375: How to get the scale factor in Monocular ORB-SLAM?

Zhujiazhen, 2017-07-07:
I am trying to run monocular ORB-SLAM with a preloaded map of the same area and if it recognizes the place it should merge the maps together. Since monocular slam can only determine the pose up to a certain scale, ORB-SLAM gives each time different result (different scale), which affects the comparison with the preloaded map.  Is there any possibility to change the scale factor of one map according to another map? Does anyone here has experience with map merging using ORB-SLAM? I appreciate any kind of help!

AlejandroSilvestri, 2017-07-08:
@Zhujiazhen 

LoopClosing recongnizes already visited places, and adjust their position and scale automatically.  If you have time, you can learn its way.

In order to merge an already saved map with a new one, not already created, I begin looking for actual frame relocalization in the loaded map, and then continue with conventional mapping.

Zhujiazhen, 2017-07-08:
@AlejandroSilvestri 

first of all thank you for reply. There is something I don't understand, in my opinion LoopClosing only recognizes visited places perfectly if we use only one map. If we use two maps , and the scale is too different, how does it know that we are at a visited place but not in a new place where everything looks the same but the scale is different? Does frame relocalization recognize places in a different way than LoopClosing?

AlejandroSilvestri, 2017-07-09:
@Zhujiazhen 
orb-slam2 doesn't save nor load maps.  It has only one map in memory, the world map, it progressively constructs starting in initialization.

If you can start with a map loaded in memory, orb-slam2 should start in relocalization mode.

When you are ready to start a new map you want to link with the map already in memory, you can try to relocalize, so the "new" map will turn into an extension to the already in memory one.

And yes, relocalization and loop closing are slightly different, because the latter has 3D point for both the actual keyframe and the candidate for loop closing; and the former only has 3D point for the candidate, but not for the actual frame.

Zhujiazhen, 2017-07-09:
@AlejandroSilvestri 
I did my own extensions for ORB slam, such that it can load a map in background. So when I run it, it loads a map, and starts to track and map just as usual, not in localizationOnly mode, and compares each keyframe against the loaded map, tries to find an "already visited place", if yes it tries to merge the loaded map with the current map. 

I am not sure how ORB slam deals with maps of different scales. Using LoopClosing::DetectLoop(), LoopClosing::ComputeSim3() (or maybe Tracking::Relocalization() also works),  it is able to recognize the place in the loaded map, but when trying to merge them (similar principle as the function CorrectLoop()), it might fail due to different scale.

Zhujiazhen, 2017-07-09:
@AlejandroSilvestri 

Could you please imagine the following situation: 
I divide one map into two parts -> submap A and submap B, and these submaps have an intersection area. Submap A is saved in the memory. We run ORB slam and build the current map which is equal to submap B. Simultaneously we compare each keyframe of map B with map A. When we enter the intersection part, it should be detected, that we are at an "already visited place" since Submap A also contains this part. Then we merge map A and map B. 

Map A was built with a certain scale which we cannot change anymore. If the current map (map B) is built with a different scale, then we cannot simply align map A to map B. One possible solution is, first assume that they are the same place regardless of the different scale (which is in general not correct), and then adjust the scale of the current map to the loaded map (But I don't know how to get this scale factor in ORB slam). 

Do you have any idea how to solve this problem? (or do I have any mistake such that this problem does not exist at all...?) 

Again thank you so much for taking time on my questions, this is really a nice help.

AlejandroSilvestri, 2017-07-09:
@Zhujiazhen 

Well, loop closing actually works on 7 degree of freedom (7DOF), 6 for pose and an extra one for scale.  It not only aligns the keyframe, also scales it.  You can study and reuse the methods LoopClosing use to do that.

You'll encounter another problem: maps suffer drifting, so they aren't euclidean.  You can align one 
keyframe, but what about the rest?  Superposed map areas could remain misaligned.   If you can identify alignment keyframes you can correct the whole map in a similar way loop closing does.

ORB-SLAM2 is a real time tool.  I believe there are many ways to align maps with not real time algorithms, not included in orb-slam2.

Zhujiazhen, 2017-07-11:
@AlejandroSilvestri 

I have studied the papers and the code for loop closing, in fact it works on 7 DOF, so this actually solves my problem. I was confused when it says that monocular slam determines the map up to a certain scale, which actually means that the whole resulted map is only correct up to a certain scale, but when having two keyframes and performing loopClosing, it is able to adjust the scale and align them (I first ignore the drifting problem). 
Thank you so much for your help!

closed 2017-07-11 by Zhujiazhen

saimanoj18, 2018-08-20:
Hi Both,

I did not read the paper, but I will be doing it soon. I have a question to ask about scale estimation.

Lets say I create a 3D point feature Map using just a monocular camera with ORB-SLAM 2, with many loop closures, then 

1. Is the map correct to metric scale. Will the estimated 3D point features in the map, really represent some keypoints extracted from real world 3D structures.
2. Based on this will I be able to load this built map and relocalize in 3D and precisely know my current position in 6D space, both 3D translation and 3D orientation ?

Thank you very much,
Sai

AlejandroSilvestri, 2018-08-20:
@saimanoj18 ,

For one, You need to build by hand a transformation matrix that can convert orb-slam2 map points coordinates from its arbitrary "world" reference, to your real world metric reference.  Keep in mind that each time you initialize orb-slam2, the initialization process define an arbitrary world map reference for the map that will be mapping.

Furthermore, the map will suffer for drifting, so the far your are from the point were you defined your transformation, the more will be the deviation.  Drifting happens in translation, scale and orientation.  Loop closing helps to minimize this effects in the loop.

Perhaps it's a new feature I'm not aware, but I don't remember orb-slam2 could load a map.

saimanoj18, 2018-08-21:
Thank you so much for your reply!

saimanoj18, 2018-08-21:
@AlejandroSilvestri 

I see sometimes in papers that people present results of ORB-SLAM2 without loop closure to show the scale drifting.

For example: https://drive.google.com/file/d/0B73o7D_54u1LcFRPeWlIR1VubzA/view Please have a look at the figure on first page.

Does it mean that with loop closure, the scale drifting would be corrected and true scale would be estimated ?

My understanding says that, scale drifting would be corrected but actually scale will not be estimated.

Could you please throw some light and correct me if I am wrong.

AlejandroSilvestri, 2018-08-21:
@saimanoj18 ,

Yes, you are correct.  Loop closure attenuates drifting, but actual scale cannot be determined by monocular systems by its own.