raulmur/ORB_SLAM2 #376: how to generate map constructed video

xuqiang, 2017-07-10:
https://www.youtube.com/watch?v=51NQvg5n-FE 

anyone know how to generate this video ?

[image]

paulsammut, 2017-07-11:
That is the KITTI 01 dataset. There is a step by step instruction on how to get it running on the readme page of this repo.

xuqiang, 2017-07-14:
@paulsammut can we save this video in file?

Alkaid-Benetnash, 2017-07-14:
@xuqiang 
As paulsammut said, you need to get ORB_SLAM2 running on the dataset first.
There will be a Viewer window to show both original input video stream(top half of your desired video) and reconstructed map(bottom half). Use any screen recorder can export the reconstructing procedure to video file.