raulmur/ORB_SLAM2 #379: Difficulties with Monocular Examples

Gasethata, 2017-07-12:
I have been trying the Monocular Examples with the TUM Dataset.
I think I did everything according to the instructions described under the TUM Dataset, and below is what I get without any output...
I'm new with SLAM and all of it so any help to proceed from here will be highly appreciated.

vip@VIPWS1:~$ cd ORB_SLAM2
vip@VIPWS1:~/ORB_SLAM2$ chmod +x build.sh
vip@VIPWS1:~/ORB_SLAM2$ ./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vip/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vip/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vip/ORB_SLAM2/build
[ 62%] Built target ORB_SLAM2
[ 68%] Built target stereo_kitti
[ 75%] Built target mono_euroc
[ 81%] Built target mono_kitti
[ 87%] Built target mono_tum
[ 93%] Built target rgbd_tum
[100%] Built target stereo_euroc
vip@VIPWS1:~/ORB_SLAM2$ ./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml/home/vip/ORB_SLAM2/rgbd-dataset_freiburg1_room-rgb.avi

Usage: ./mono_tum path_to_vocabulary path_to_settings path_to_sequence
vip@VIPWS1:~/ORB_SLAM2$

Alkaid-Benetnash, 2017-07-14:
Hi, you downloaded the wrong dataset.
[image]
you should follow the **tgz** link and download an archive file. Then you should uncompress it to a directory.
Finally, the third parameter you pass to mono_tum should be a directory with the following structure.
```
path_to_sequence
├── accelerometer.txt
├── depth
├── depth.txt
├── groundtruth.txt
├── rgb
└── rgb.txt
```

Gasethata, 2017-07-17:
Thanks a lot @Alkaid-Benetnash