raulmur/ORB_SLAM2 #383: Sparcity in Local mapping

anerwolf, 2017-07-17:
Hi,

I am collecting statistics on the flow of the application with the example data sets. i have noticed that the matrices that are solved in local BA (in Local mapping) are not sparced at all. is that right or am i looking at wrong place? i am talking about the solution to the landmarks. if i am correct, why do you implement the algorithm with sparced methods?

Thanks,
Aner

AlejandroSilvestri, 2017-07-17:
@anerwolf 

Your finding is a good observation, perhaps can lead to an improvement.

Sparsity in local map highly depends on scene.  It should be analysed in many different scenarios.  May be someone will find g2o::SparseOptimizer is not the most suitable for a general application local map BA.

anerwolf, 2017-07-18:
@AlejandroSilvestri 
I meant to say i am looking at the poses solution process. 
I dont know if it is dependent on the scene since in each of the datasets i get a full matrix to solve (100% nnz) for the poses, without any exceptions. i thought maybe it is due to the fact that the subset of key frames you collect for the local mapping are attached with regards to the covisabiltiy graph. are there any other data sets (except from the examples one) i can use that will be compatible for the ORB-SLAM2?
Thanks,
Aner

AlejandroSilvestri, 2017-07-18:
@anerwolf 

Just to clarify: are you working with local bundle adjustment (in local mapping thread), or pose optimization (in main thread)?

Both use g2o::SparseOptimizer, and I believe neither of both have sparse matrix.

anerwolf, 2017-07-19:
@AlejandroSilvestri 

I am working with local BA (in local mapping thread).
Why would you use all the sparse matrix framework (CCS, SparceBlockMatrix, etc.) if the most working thread (local mapping) doesnt have a sparse matrix to begin with?

Thanks,
Aner

AlejandroSilvestri, 2017-07-20:
@anerwolf 

SLAM is a general problem posed with a sparse matrix.

Where are you taken the matrix from?

anerwolf, 2017-07-20:
@AlejandroSilvestri
I have taken the matrix from the function fillSparseMatrix at linear_solver_eigen.h file located at the g2o.
it receives a block matrix and creates triplets of (i,j,val) so i just counted the nnz elements and extracted the size of the matrix.