raulmur/ORB_SLAM2 #397: Question about running real-time camera via ROS

baike960, 2017-07-27:
Hi,
I am new to SLAM and I have been trying to run this project. I succeeded in running the datasets for Monocular, binocular and RGBD, but when I tried to run the ROS example, I did the following:
roscoe
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/parallels/NewORB/ORB_SLAM2/Examples/ROS
rosrun ORB_SLAM2 Mono /home/parallels/NewORB/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/parallels/NewORB/ORB_SLAM2/Examples/Monocular/TUM1.yaml

And there was a black screen with "wait for images". I know I probably need to adjust the path of settings file and vocabulary file, so could someone tell me how to connect a real-time camera to this project? I will probably test with a simple usb monocular camera and then a Kinect for xbox one on Ubuntu 16.04.

Thanks in advance!

laxnpander, 2017-07-31:
Ros mono node expects images to be published on /camera/image_raw. Therefore you will need a camera node that reads images to your buffer and publishes them. Additionally you will need to calibrate your camera, because TUM1.yaml is most likely not for your specific camera.

jtbon20, 2017-08-08:
@baike960 
After you launch your camera, run `rostopic list` to find what topics your camera publishes to.
After you find some that sound reasonable run `rviz` or `rosrun image_view image_view image:=<image topic> [image transport type]` to check the output and finalize which topic you're going to use
Finally when running orbslam add ` /camera/image_raw:={the/path/to/your/chosen/topic}` to the end of the command