raulmur/ORB_SLAM2 #399: Failed to build ORB_SLAM on UBUNTU 16.04

pulkitver1991, 2017-07-28:
config.h not found!

Scanning dependencies of target rospack_genmsg_libexe
[  0%] Built target rospack_genmsg_libexe
Scanning dependencies of target rosbuild_precompile
[  0%] Built target rosbuild_precompile
Scanning dependencies of target ORB_SLAM
[  5%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/main.cc.o
In file included from /home/pulkit/ORB_SLAM/ORB_SLAM/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:37:0,
                 from /home/pulkit/ORB_SLAM/ORB_SLAM/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/pulkit/ORB_SLAM/ORB_SLAM/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/pulkit/ORB_SLAM/ORB_SLAM/include/LoopClosing.h:33,
                 from /home/pulkit/ORB_SLAM/ORB_SLAM/include/LocalMapping.h:26,
                 from /home/pulkit/ORB_SLAM/ORB_SLAM/include/Tracking.h:31,
                 from /home/pulkit/ORB_SLAM/ORB_SLAM/src/main.cc:29:
/home/pulkit/ORB_SLAM/ORB_SLAM/Thirdparty/g2o/g2o/types/../core/openmp_mutex.h:30:26: fatal error: ../../config.h: No such file or directory
compilation terminated.
CMakeFiles/ORB_SLAM.dir/build.make:118: recipe for target 'CMakeFiles/ORB_SLAM.dir/src/main.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM.dir/src/main.cc.o] Error 1
CMakeFiles/Makefile2:227: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Kindly Suggest ?

Santi-7, 2017-07-28:
Have you executed build.sh?

EugeneLew, 2017-07-31:
Try adding the `Thirdparty` folder or `Thirdparty/g2o` folder into the include directories list in the `CMakeLists.txt`

Pratapabhay, 2017-07-31:
Did you able to run it? @pulkitver1991

pulkitver1991, 2017-09-18:
@EugeneLew 
Thanks for your response, and sorry for beong late. I solved the issue using the same idea..Mark this issue as Closed. Thanks