raulmur/ORB_SLAM2 #400: Raspberry Pi 3 Compile Problems

ryutatsu, 2017-07-28:
I'm encountering a couple of errors:

double free or corruption errors and c++ doesn't support c++11 for some reason.

```bash
pi@raspberrypi:~/workspace/ORB_SLAM2 $ sudo ./build.sh
Configuring and building Thirdparty/DBoW2 ...
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/workspace/ORB_SLAM2/Thirdparty/DBoW2/build
Scanning dependencies of target DBoW2
[ 14%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/BowVector.cpp.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x01f10e38 ***
CMakeFiles/DBoW2.dir/build.make:62: recipe for target 'CMakeFiles/DBoW2.dir/DBoW2/BowVector.cpp.o' failed
make[2]: *** [CMakeFiles/DBoW2.dir/DBoW2/BowVector.cpp.o] Aborted
make[2]: *** Waiting for unfinished jobs....
[ 28%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o
[ 42%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FeatureVector.cpp.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x00434e38 ***
CMakeFiles/DBoW2.dir/build.make:110: recipe for target 'CMakeFiles/DBoW2.dir/DBoW2/FeatureVector.cpp.o' failed
make[2]: *** [CMakeFiles/DBoW2.dir/DBoW2/FeatureVector.cpp.o] Aborted
[ 57%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/ScoringObject.cpp.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x008a6e38 ***
CMakeFiles/DBoW2.dir/build.make:86: recipe for target 'CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o' failed
make[2]: *** [CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o] Aborted
*** Error in `/usr/bin/c++': double free or corruption (top): 0x01236e38 ***
CMakeFiles/DBoW2.dir/build.make:134: recipe for target 'CMakeFiles/DBoW2.dir/DBoW2/ScoringObject.cpp.o' failed
make[2]: *** [CMakeFiles/DBoW2.dir/DBoW2/ScoringObject.cpp.o] Aborted
[ 85%] Building CXX object CMakeFiles/DBoW2.dir/DUtils/Random.cpp.o
[ 85%] Building CXX object CMakeFiles/DBoW2.dir/DUtils/Timestamp.cpp.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x01220e38 ***
CMakeFiles/DBoW2.dir/build.make:158: recipe for target 'CMakeFiles/DBoW2.dir/DUtils/Random.cpp.o' failed
make[2]: *** [CMakeFiles/DBoW2.dir/DUtils/Random.cpp.o] Aborted
*** Error in `/usr/bin/c++': double free or corruption (top): 0x00252e38 ***
CMakeFiles/DBoW2.dir/build.make:182: recipe for target 'CMakeFiles/DBoW2.dir/DUtils/Timestamp.cpp.o' failed
make[2]: *** [CMakeFiles/DBoW2.dir/DUtils/Timestamp.cpp.o] Aborted
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/DBoW2.dir/all' failed
make[1]: *** [CMakeFiles/DBoW2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Configuring and building Thirdparty/g2o ...
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- BUILD TYPE:Release
[... 134 lines cut ...]

(gedit:29975): Gtk-WARNING **: Calling Inhibit failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.SessionManager was not provided by any .service files
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
*** Error in `/usr/bin/c++': double free or corruption (top): 0x01fcc8f0 ***
-- Performing Test COMPILER_SUPPORTS_CXX11 - Failed
-- Performing Test COMPILER_SUPPORTS_CXX0X
*** Error in `/usr/bin/c++': double free or corruption (top): 0x018f88f0 ***
-- Performing Test COMPILER_SUPPORTS_CXX0X - Failed
CMake Error at CMakeLists.txt:26 (message):
  The compiler /usr/bin/c++ has no C++11 support.  Please use a different C++
  compiler.

-- Configuring incomplete, errors occurred!
See also "/home/pi/workspace/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/workspace/ORB_SLAM2/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
```

jyl58, 2017-07-29:
me too.
I deleted the "-march=native" in CMakeLists.txt  ,solved this problem .

ryutatsu, 2017-07-31:
unfortunately this didn't work, I don't have the output as the PI ended up crashing and rebooting during the compilation.

jyl58, 2017-08-01:
Did you delete all "-march=native" in DBoW , G2O and ORB_SLAM's CMakeLists.txt ?

EugeneLew, 2017-08-01:
Try running the `make` command with `-j1` flag, i.e. `make -j1`, instead of the original `make -j`. (Change it in the `build.sh`)

ryutatsu, 2017-08-01:
removing `-march=native` removal and changing to `-j` seems to have worked, thanks.

YuehChuan, 2017-09-16:
@ryutatsu  cool! I wonder  how was the performance on pi3, the frame rate and what camera you use?
TQ!