raulmur/ORB_SLAM2 #402: Error running ROS Mono

rosbeginneruser, 2017-08-01:
Built and ran properly the example datasets, but when I tried to run it with ROS ran into a few problems. Seeking help -- other similar issues said to "After removing gtk+ 2.x, reinstalling Opencv and ROS, everything works well. Make sure that you are compiling the codes (cmake projects) using the same gtk version!" @David-Willo in [issue 325](https://github.com/raulmur/ORB_SLAM2/issues/325) But I had a few questions:

Shouldn't I not remove gtk+ 2.x due to its necessity in the system?
How can I compile with cmake to ensure that I'm using the same gtk version?

Terminal output:

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 535.4
- fy: 539.2
- cx: 320.1
- cy: 247.6
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 100
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

(Mono:9555): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
Trace/breakpoint trap (core dumped)

rosbeginneruser, 2017-08-01:
Update: Redid OpenCV with cmake GTK ON and now have a similar error but new number:

(Mono:27720): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
Trace/breakpoint trap (core dumped)

rimitalahiri, 2017-08-08:
@rosbeginneruser could you solve the problem??? I am also facing the same and posted an issue. If you have got rid of the issue, please share the solution with us..

jtbon20, 2017-08-08:
@rimitalahiri uninstall opencv then remake with cmake with your favorite flags and add `-D WITH_GTK_2_X=ON`
make sure to install necessary linbraries `sudo apt install libgtk2.0-dev pkg-config`
and then rebuild ROS and reinstall ORB SLAM

Santi-7, 2017-08-19:
I experimented the same issue, and @jtbon20 answer solved it.

benjamintanweihao, 2018-12-15:
Had the same problem. I just had to recompile with `-D WITH_GTK_2_X=ON` OpenCV and reinstall and that was it.