raulmur/ORB_SLAM2 #403: Error in building ORB_SLAM2 in ROS Kinetic version

rimitalahiri, 2017-08-02:
Following the tutorial provided in this [link](https://github.com/raulmur/ORB_SLAM2), I am trying to run ORB_SLAM as a ROS node in ROS Indigo. After executing the line /.build_ros.sh, I am facing a list of errors,
```
/home/ubuntu_16/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30:22: fatal error: Eigen/Core: No such file or directory
compilation terminated.
CMakeFiles/RGBD.dir/build.make:118: recipe for target 'CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o' failed
make[2]: *** [CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2

```
I am unable to get rid of these problem, if someone has faced the same, please help. Thank you.

jtbon20, 2017-08-02:
@rimitalahiri are you trying with Indigo or Kinetic?

rimitalahiri, 2017-08-02:
@jtbon20 I am trying with ROS Kinetic and Ubuntu 16.04.

EugeneLew, 2017-08-02:
Have you put `eigen3` folder into `/usr/local/local`? You don't have to install eigen. If you have done that, and it still doesn't work, put `/usr/local/local/eigen3` into the include directories in the `CMakeLists.txt`. This tells CMake where to find `Eigen/Core`.

rimitalahiri, 2017-08-03:
This problem is solved, by simply this line of code...
sudo ln -s /usr/include/eigen3/Eigen /usr/local/include/Eigen

closed 2017-08-03 by rimitalahiri

hhyh, 2018-03-07:
Below change in Examples/ROS/ORB_SLAM2/CMakeLists.txt can resolve the issue

 include_directories(
 ${PROJECT_SOURCE_DIR}
 ${PROJECT_SOURCE_DIR}/../../../
 ${PROJECT_SOURCE_DIR}/../../../include
 ${Pangolin_INCLUDE_DIRS}
+${EIGEN3_INCLUDE_DIR}
 )

ujjwalrathod007, 2018-07-04:
@hhyh @rimitalahiri both kind of worked!! thank you