raulmur/ORB_SLAM2 #404: (core dumped) error with gtk while running ORB_SLAM2 in ROS

rimitalahiri, 2017-08-03:
I am running ORB_SLAM2 in ROS Kinetic as suggested in this [link](https://github.com/raulmur/ORB_SLAM2). After executing the line 
rosrun ORB_SLAM2 Mono /home/ubuntu_16/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/ubuntu_16/ORB_SLAM2/Examples/Monocular/TUM1.yaml,
I am getting this output,
```
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.
(Mono:7626): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
Trace/breakpoint trap (core dumped)
```
I am not understanding what is the problem with gtk 2 and 3, please help if someone has faced similar issues.
Thank you.

wangjiefjj, 2018-03-02:
Did you have solved this issue?

cisnow, 2019-09-26:
> Did you have solved this issue?

daxiongdi，Did you have solved this issue?

xujinjinjinjin, 2020-12-18:
hi  Did you have solved this issue?

xujinjinjinjin, 2020-12-20:
> hi Did you have solved this issue?

I bypassed it by going in the g2o cmakelists and the orb slam cmakelists and removing -march=native .yeah~