raulmur/ORB_SLAM2 #405: Fail to compile ORB-SLAM2

balzar29, 2017-08-03:
Hi, this is the error:

`make[2]: *** [CMakeFiles/ORB_SLAM2.dir/build.make:495: CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/build.make:63: CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:253: CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`
I'm on ArchLinux not Ubuntu, maybe this could be the problem, but i've installed all the dependences.

EugeneLew, 2017-08-04:
I could be wrong, but I think there should be something else indicating the exact error, before these lines. Try running the `make` command with `make -j1` and see before you can get the error before these lines.

balzar29, 2017-08-04:
thanks for answer! I found something like this:

`/home/io/Programmi/ORB_SLAM2-master/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackStereo(const cv::Mat&, const cv::Mat&, const double&)’:
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:134:17: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                 ^~~~~~
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:134:17: note: suggested alternative: ‘fseek’
                 usleep(1000);
                 ^~~~~~
                 fseek
/home/io/Programmi/ORB_SLAM2-master/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackRGBD(const cv::Mat&, const cv::Mat&, const double&)’:
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:185:17: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                 ^~~~~~
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:185:17: note: suggested alternative: ‘fseek’
                 usleep(1000);
                 ^~~~~~
                 fseek
/home/io/Programmi/ORB_SLAM2-master/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackMonocular(const cv::Mat&, const double&)’:
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:236:17: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                 ^~~~~~
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:236:17: note: suggested alternative: ‘fseek’
                 usleep(1000);
                 ^~~~~~
                 fseek
/home/io/Programmi/ORB_SLAM2-master/src/System.cc: In member function ‘void ORB_SLAM2::System::Shutdown()’:
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:309:13: error: ‘usleep’ was not declared in this scope
             usleep(5000);
             ^~~~~~
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:309:13: note: suggested alternative: ‘fseek’
             usleep(5000);
             ^~~~~~
             fseek
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:315:9: error: ‘usleep’ was not declared in this scope
         usleep(5000);
         ^~~~~~
/home/io/Programmi/ORB_SLAM2-master/src/System.cc:315:9: note: suggested alternative: ‘fseek’
         usleep(5000);
         ^~~~~~
         fseek
`

Santi-7, 2017-08-04:
See #144

You have to `include <unistd.h>`. This PR includes it in several files. You can just include it in `System.h`.

siddu1998, 2019-01-30:
this is due to the sleep function, choices you have to solve this is
a. add #include <unistd.h> in all those files where usleep error pops up 
b. Navigate to the System.h folder and add #include <unistd.h>