raulmur/ORB_SLAM2 #406: Ask for reduced light Vocabulary file!

realitian, 2017-08-07:
By https://github.com/raulmur/ORB_SLAM2/issues/82, I know that the recreated vocabulary of 5 levels can reduce the memory usage extremely.
Can anybody help me by sharing the recreated vocabulary of 5 levels instead of 6?
Thank in advance.

wangzheqie, 2017-08-10:
Yet. I create a very small vocabulary , and it still work well, and here is my codes to create voc.txt
https://github.com/wangzheqie/create_DBoW2_voc.git

realitian, 2017-08-10:
Hi @wangzheqie 
Thanks for share that.
Btw, how many did you trained images for that voc?
Because it does not work on my side, the system has lost tracking occasionally.
Cheers.

AlejandroSilvestri, 2017-08-15:
@realitian 

I believe the reduced vocabulary is a subset of the original vocabulary.  It isn't a new trained one.

Less voc levels demands more matching work in both tracking and mapping.  How much, I don't know.

realitian, 2017-08-16:
@AlejandroSilvestri 
Thanks for reply.
I had trained again with 2008.09.01 imageset, and got around 4M voc.bin, already.
it works well.

xzwang12345, 2021-04-01:
Hi@wangzheqie @Realitian,I can't get the bovisa-2008-09-01 dataset from http://www.rawseeds.org/rs/capture_sessions/view/10 due to that this torrent seem like out of source.If you have this dataset, can you share it? Thanks!