raulmur/ORB_SLAM2 #410: Augmented reality with ORB-SLAM2(Monocular)

prathamsatam, 2017-08-10:
Augmented reality with ORB-SLAM2(Monocular) - this is a sample video on github 
link-https://www.youtube.com/embed/T-9PYCKhDLM

how to achieve the same output using ORB-SLAM2? Which dataset I have to use to implement? and what is the requirement and process to do the same?

Thank you.

AlejandroSilvestri, 2017-08-15:
@prathamsatam 

You must compile mono_ros_AR.cc , and then provide images via ROS.  That video is not a dataset, is a real time video.

yizhou-wang, 2018-08-08:
@AlejandroSilvestri Is there a way to use some exist datasets like KITTI for monoAR's input? Thank you!

AlejandroSilvestri, 2018-08-08:
@yizhou-wang , I'm not using datasets, but I think you can provide datasets images via ROS too.

yizhou-wang, 2018-08-15:
@AlejandroSilvestri Thanks a lot! But I met another problem: I could not find out how does ORB-SLAM2 monoAR demo get the point to put cube? In the AR demo, the place to put cube should be selected by the users, but I want to try a fixed point. Is there a way to do that? Thank you for your help!

AlejandroSilvestri, 2018-08-15:
@yizhou-wang , all you need to put a cube in a fixed point is its pose in the "world" reference system (the one orb-slam2's map uses).  And code it.

To this end you can learn the code starting [here](https://github.com/raulmur/ORB_SLAM2/blob/master/Examples/ROS/ORB_SLAM2/src/AR/ViewerAR.cc#L336).

yizhou-wang, 2018-08-15:
@AlejandroSilvestri As far as my understanding, the monoAR draws the cube at [this line](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/Examples/ROS/ORB_SLAM2/src/AR/ViewerAR.cc#L215). The function `DrawCube` is declared as 
```
void DrawCube(const float &size, const float x=0, const float y=0, const float z=0);
```
and `(x,y,z)` should be the place to put cube? But there is no input of `(x,y,z)` so they always use the default value `(0,0,0)`. However, how does it know where did the user click? 

I'm new to ORB-SLAM and thank you so much for your patience!