raulmur/ORB_SLAM2 #412: Identification of changes between map and current environment in tracking only mode

rosbeginneruser, 2017-08-11:
@raulmur @AlejandroSilvestri @wangzheqie After it's set into Localization only mode, how can I tell if there are differences between the map and the current environment in terms of new KeyPoints or points that it would add were it still in mapping mode?

jtbon20, 2017-08-11:
Correct me if I'm wrong, @AlejandroSilvestri or @raulmur but it would involve comparing map points from keyframe given by the covisibility graph after localization with current sensor map points ??

AlejandroSilvestri, 2017-08-12:
@rosbeginneruser , @jtbon20 ,

There may be many ways to do that, none of them trivial.  In tracking only mode there's no intent to add new map points.

If you manage to identify new map points, they can point out differences with map.

Each map point has two properties, recording the times the point should be observed, and the times the point was observed.  You can tell when a map points should be observed but it doesn't.