raulmur/ORB_SLAM2 #414: what():  Pangolin X11: Failed to create an OpenGL context

JiankaiSun-SJTU-MVIG-training, 2017-08-17:
I meet an error when I run the latest code of ORB_SLAM2 on server.
The system is 
- Ubuntu 14.04 64-bit
- OpenCV 2.4.8

The error information:
```
jack@gpu02:/SSD2/jack/Applications/ORB_SLAM2$ ./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml Examples/rgbd_dataset_freiburg1_xyz

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

modprobe: ERROR: ../libkmod/libkmod-module.c:809 kmod_module_insert_module() could not find module by name='fglrx'
modprobe: ERROR: could not insert 'fglrx': Function not implemented
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: ../libkmod/libkmod-module.c:809 kmod_module_insert_module() could not find module by name='fglrx'
modprobe: ERROR: could not insert 'fglrx': Function not implemented
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: ../libkmod/libkmod-module.c:809 kmod_module_insert_module() could not find module by name='fglrx'
modprobe: ERROR: could not insert 'fglrx': Function not implemented
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

X11 Error: BadValue (integer parameter out of range for operation)
X11 Error: BadValue (integer parameter out of range for operation)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to create an OpenGL context
Aborted (core dumped)
```

ramanpreet9, 2017-10-10:
I am also getting the same error

geoffviola, 2017-10-23:
I had a similar issue. Try typing ```glxinfo```, which is part of mesa-utils. If you have an nvidia graphics card try ```nvidia-smi```. Both of those programs gave strange errors so it seemed to be a driver issue for me. Anyway, restarting the computer seemed to have fixed the issue.

SimoGira, 2017-11-08:
Same problem! If I run directly on the server it works perfectly, but if I run it in remote, it doesn't work.

geoffviola, 2017-11-08:
@SimoGira silly question, but how are you running it remotely. Can you run any graphical windows remotely? If you're using ssh, are you passing the "-X" parameter

SimoGira, 2017-11-08:
Yes, I’m using ssh with -X and yes I’m able to running any windows on my host (e.g gedit etc..), but with orbslam2 still not work

zezuladp, 2018-01-23:
I got around this by removing Pangolin dependencies after debugging via visualization was no longer needed.
https://gist.github.com/zezuladp/165d3c9b91ec936f931b5d5b5df74f3a
`git apply system.patch` from root dir.

There's also some ways to use Nvidia settings and configure a virtual display
some info on this [here](http://jasonjuang.blogspot.com/2016/02/how-to-configure-xorgconf-for-headless.html)

gravitychen, 2019-06-01:
if you are windows, try to use mobaXterm to connect your server , and use the same command ,
than it works

grafoteka, 2019-07-23:
> I got around this by removing Pangolin dependencies after debugging via visualization was no longer needed.
> https://gist.github.com/zezuladp/165d3c9b91ec936f931b5d5b5df74f3a
> `git apply system.patch` from root dir and then rebuilding `./build.sh`
> 
> There's also some ways to use Nvidia settings and configure a virtual display
> some info on this [here](http://jasonjuang.blogspot.com/2016/02/how-to-configure-xorgconf-for-headless.html)

Thank you @zezuladp. It work for me.