raulmur/ORB_SLAM2 #415: why the localization model int the same place at different angles  will track the loss?

zgq91, 2017-08-18:
Hi:
   when I have a good map int the work area,  but the same area ,the tracking of different angles will be lost.
why?   whether the condition in the relocation function  is too strict.

AlejandroSilvestri, 2017-08-20:
@zgq91 
What do you mean by different angle?  A different point of view, or a rotated camera looking at the same direction?

zgq91, 2017-08-23:
@AlejandroSilvestri  a rotated camera  looking at the same direction

AlejandroSilvestri, 2017-08-24:
@zgq91 , ok, you are rotating the camera around its Z axis (its viewing direction).

BRIEF descriptors aren't good recognizing rotated patterns.  They are not robust to rotation at all.  And because you remain at the same location, no new keyframes are added to progressively record new descriptors.

Common robot application doesn't rotate in that way.  I believe this problem can be overcome by recording the map points descriptors from frames during the rotation.  Not as easy as it sounds.

AlejandroSilvestri, 2017-08-25:
@zgq91 , that's the problem with visual SLAM, you don't get absolute coordinates.  You must provide them outside orb-slam2.

AR demo has a dominant plane detector, which can be used to detect the ground.

lizzzcai, 2017-08-26:
for me the plane detector only get the pose of the plane , not include the location.  Do you know how to solve it.

AlejandroSilvestri, 2017-08-26:
@lizzzcai 

The algorithm I pointed out get the dominant plane from a set of map points.  You can use those map points to get their location.