raulmur/ORB_SLAM2 #417: Error while building ORB_SLAM2

rimitalahiri, 2017-08-23:
I am getting the following error after executing /.build.sh.

CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find Eigen3 (missing: EIGEN3_INCLUDE_DIR EIGEN3_VERSION_OK)
  (Required is at least version "3.1.0")
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindEigen3.cmake:82 (find_package_handle_standard_args)
  CMakeLists.txt:70 (FIND_PACKAGE)

How to get rid of this issue.
Thank you.

wenhaoneu, 2017-09-08:
Have you configured the eigen dependencies yet? If not, try to run the command "sudo apt-get install libeigen3-dev" first.

bdjohnson529, 2017-12-20:
My experience with ORBSLAM and eigen:
-Download Eigen **version 3.2** from [the website](http://eigen.tuxfamily.org/index.php?title=Main_Page). The latest version 3.3 did not work for me.
-Decompress the folder. Open the INSTALL file for installation instructions. Installation is typical of CMake, despite what is advertised on the Eigen website. I'm sure you could also manually move the header files to /usr/local/include/ or similar, but honestly CMake is much simpler.

Hope this helps.