raulmur/ORB_SLAM2 #418: QObject::~QObject: Timers cannot be stopped from another thread Segmentation fault (core dumped)

JiankaiSun-SJTU-MVIG-training, 2017-08-24:
From [Alkaid-Benetnash/ORB_SLAM2](https://github.com/Alkaid-Benetnash/ORB_SLAM2), After changes some code of the original `mono_kitti.cc`  to
```
if(argc != 5)
    {
        cerr << endl << "Usage: ./mono_kitti path_to_vocabulary path_to_settings path_to_sequence [1|0](save map?)" << endl;
        return 1;
    }

    // Retrieve paths to images
    vector<string> vstrImageFilenames;
    vector<double> vTimestamps;
    LoadImages(string(argv[3]), vstrImageFilenames, vTimestamps);

    int nImages = vstrImageFilenames.size();

    // Create SLAM system. It initializes all system threads and gets ready to process frames.
    ORB_SLAM2::System SLAM(argv[1],argv[2],ORB_SLAM2::System::MONOCULAR,true,(bool)atoi(argv[4]));
```
The above code pass the compile. But I met the following error when I run `mono_kitti`. Do you have any idea about it? Thank you!

```
$ ./Examples/Monocular/mono_kitti Vocabulary/ORBvoc.txt Examples/Monocular/KITTI00-02.yaml Examples/dataset/sequences/00 1

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Cannot Open Mapfile: map.bin , Create a new one

Camera Parameters: 
- fx: 718.856
- fy: 718.856
- cx: 607.193
- cy: 185.216
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 10
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 4541

New Map created with 158 points
Loop detected!
Local Mapping STOP
Local Mapping RELEASE
Starting Global Bundle Adjustment
Global Bundle Adjustment finished
Updating map ...
Local Mapping STOP
Local Mapping RELEASE
Map updated!
Loop detected!
Local Mapping STOP
Local Mapping RELEASE
Starting Global Bundle Adjustment
Global Bundle Adjustment finished
Updating map ...
Local Mapping STOP
Local Mapping RELEASE
Map updated!
Saving Mapfile: map.bin ...done
-------

median tracking time: 0.0548361
mean tracking time: 0.0597198

Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!
QObject::~QObject: Timers cannot be stopped from another thread
Segmentation fault (core dumped)

```

My system:
- OpenCV 2.4.13.3
- Ubuntu 16.04 64-bit

rimitalahiri, 2017-08-24:
I am also facing same issue, I have not changed anything in the code though. Please suggest a solution.

rimitalahiri, 2017-08-25:
@JiankaiSun-SJTU-MVIG-training did you find any solution????

JiankaiSun-SJTU-MVIG-training, 2017-08-26:
I haven't find the solution yet. :( 
Waiting for help from power users....

Yannnnnnnnnnnn, 2017-10-19:
I also met this. But after I recompiled the opencv without Qt, everything worked well.

ichigoi7e, 2017-11-01:
@Yannnnnnnnnnnn Could I know the deatils about your operations？

yingyingqin, 2018-03-10:
I also met this problem, What's the meaning of 'recompiled the opencv without Qt'? I compile opencv by 'cmake and make', It's nothing with Qt. @Yannnnnnnnnnnn

Yannnnnnnnnnnn, 2018-03-10:
http://blog.csdn.net/u012348774/article/details/78284445 @yingyingqin @ichigoi7e 
I compiled the opencv by myself and canceld the WITH_QT option.

HITHeart, 2019-03-26:
> I also met this problem, What's the meaning of 'recompiled the opencv without Qt'? I compile opencv by 'cmake and make', It's nothing with Qt. @Yannnnnnnnnnnn

Have you installed ROS?Perhaps,the opencv in ROS compiled with QT.

azkahariz, 2019-11-13:
> http://blog.csdn.net/u012348774/article/details/78284445 @yingyingqin @ichigoi7e
> I compiled the opencv by myself and canceld the WITH_QT option.

I use QT for my other project :(
Is there any other way?