raulmur/ORB_SLAM2 #419: Change ORB to Surf features

scofield1991, 2017-08-28:
I want to use Surf  features instead of ORB. May be someone did it before? What is a pipeline for doing it?

AlejandroSilvestri, 2017-08-30:
@scofield1991 , 

Some asked something like that, anyone actually did it.  I believe it is possible, but not easy nor straightforward.  But keep in mind ORB was chosen because it is fast.  Do not expect speed from SURF.

Orb-slam2 is modular, it wrapped ORB in ORBextractor and ORBmatcher classes.

ORBextractor applies ORB: uses FAST as feature detector (should replace by SURF detector) and ORB as descriptor extractor (should replace by SURF).  Note it doesn't uses opencv's already implemented ORB, instead it implements its own slightly adapted version of ORB, dividing the image in cells prior processing FAST.

ORBmatcher::DescriptorDistance is the method to calculate distance between two descriptors.  Is works on binary descriptors, you should rewrite it to work with floating point descriptors like SURF's.

You should correct ad hoc distance thresholds all around the code, I believe they are only in ORBmatcher.cc .

You should adapt or replace BoW in order to work with SURF descriptors.

wangzheqie, 2017-09-07:
>This system detect features via **FAST**  in OpenCV which is not contain rotation information, then the **image moment** is used as **IC_Angle()** to add rotation information. But **SIFT** dose not need.

>**ORB** and **SIFT** descriptors for each detected image points are different, so need to change the comparison function like **SearchByBoW()** associated with **DBoW2**

>During **TrackLocalMap** , the system will match the **unmatched** features through **connected keyframes** , remember to check the feature match function.
 
>After the **Local Mapping** , the last processed KeyFrame may inserted into the **covisibility graph**,  and then inserted into a **recognition database** , remember to check the descriptor extract function

>##My opinion is **TRY NOT** to do that, because this may cost you lots of time and finally you may find it **NOT BETTER THAN ORB** at all.

closed 2017-12-14 by scofield1991