raulmur/ORB_SLAM2 #421: Map Save/Load with Latest ORB_SLAM2 Version

fawadahm, 2017-08-31:
Hi,

I was wondering whether there is an ORB_SLAM2 repo available with map save/load options. I have tried the versions from @Pione, @MathewDeny and the other ones out there but they only work on Ubuntu 14 with the previous versions of ORB_SLAM2.

I'm looking for a solution with the new ORB_SLAM2, on Ubuntu 16.04

I tried to tailor the current ORB_SLAM2 with map save/load code from @MathewDenny's repo but I run into the following error when trying to load the map:

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded in 0.17s
terminate called after throwing an instance of 'boost::archive::archive_exception'
  what():  input stream error

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

poine, 2017-09-01:
I have not played with orbslam in a while but i am positive that i have
used it on ubuntu 16.04
Can you give more details about your crash? I remember one of the issues
when switching from ubuntu 14 to ubuntu 16 was the version of opencv (v3 vs
v2)

fawadahm, 2017-09-05:
Thank you for your kind reply.

The issue was that I would get a SIGSEV error each time I tried to load the map. I didn't dive deep into the code so I could not exactly figure out what the exact reason for the error was.

As far as the OpenCV version was concerned, I just changed the required version in the CMake files to 3.1.0. Could that have been causing the issue?

Anyhow, I managed to find another repo that works perfectly with Ubuntu 16.04. You can find it at:

https://github.com/Alkaid-Benetnash/ORB_SLAM2

wangzheqie, 2017-09-07:
I am use OpenCV 2.4.9 and it work will.

shadowver, 2018-08-03:
this mostly like caused by the boost::archive, make sure when you save and load map, achive flag is no_header.