raulmur/ORB_SLAM2 #422: Track keyframes using a live camera

kevin-george, 2017-09-01:
I am trying to use this library with a monocular camera. And I wish to track the position of the keyframes(and possibly publish it as a ROS message).

My question is, has anyone figured out how to transform the pose which is going to be in the camera coordinate frame to the world co-ordinate frame(preferably right handed system which is used by ROS i.e. x forward, z up)

I have seen the pull request [102](https://github.com/raulmur/ORB_SLAM2/pull/102/files) but the author themselves is unsure what some of their code does.

AlejandroSilvestri, 2017-09-02:
@kevin-george ,

Actual pose is in main function:

    SLAM.GetTracker()->mCurrentFrame.mTcw 

mTcw is a Mat 4x4 pose matrix, world pose referenced to the camera.  Usually you want its inverse, camera pose in reference to the world.

You also have

    mCurrentFrame.GetRotationInverse()
    mCurrentFrame.GetCameraCenter()

which give you the rotation matrix and the position, in reference to the world.

This is a right handed system, and World is a completely arbitrary reference system, so it has no clue what "up" means.  You should make your own transform from ORB-SLAM2 World reference system to your ground reference system with Z pointing up.  World system is defined in initialization, so it will be different each time you initialize.

I'm not that familiar with ROS, could you tell me how to publish a pose?  Like what line of code you use to publish a Mat on ROS.

rnunziata, 2017-10-23:
Where is this functions GetTracker?  when I try to use mCurrentFrame it  always  results in segfault.

AlejandroSilvestri, 2017-10-23:
@rnunziata 

My mistake.  You need a way to get System::mpTracker, which is private.  You can add this method in System.h

    Tracking* System::GetTracker(){return mpTracker;}

closed 2020-02-12 by kevin-george

mmahdavian, 2020-09-14:
@AlejandroSilvestri  Thank you for you respond here. What you mentioned will publish all the frames matrices. Do you know how can we get pose or matrix of only keyframes?

Thank You

AlejandroSilvestri, 2020-09-14:
@mmahdavian,

    std::vector<KeyFrame*> Map::GetAllKeyFrames()

will get you all KeyFrames in map.  You can then use KeyFrame's method to get pose or position, like 

    GetPose()
    GetTranslation()

This is a not working example:

    Mat pose = SLAM.GetMap->GetAllKeyFrames()[i]->GetPose();