raulmur/ORB_SLAM2 #423: ubuntu17 segmentation fault ORB-SLAM2

wenhaoneu, 2017-09-02:
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 520.909
- fy: 521.007
- cx: 325.141
- cy: 249.702
- k1: 0.231222
- k2: -0.784899
- k3: 0.917205
- p1: -0.003257
- p2: -0.000105
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.07156

-------
Start processing sequence ...
Images in the sequence: 792

段错误 (核心已转储)

wangzheqie, 2017-09-07:
刚刚加载完图片进入**TRACK** 就出错，这样子信息太少，看不出是哪里出错。

wenhaoneu, 2017-09-07:
Thank you for your reply.Luckily I have known the reason.   At first, I followed the instructions on readme.txt to test the ORB-SLAM2. When I tried to run the rgbd dataset,  the problem happened.   It was because my depth images was not match with the associnations.txt, which caused the executable program coudn't find the depth image correctly. The program had access to null pointers and caused segmentation fault. -------- 原始邮件 --------主题：Re: [raulmur/ORB_SLAM2] ubuntu17 segmentation fault ORB-SLAM2 (#423)发件人：wangzheqie 收件人：raulmur/ORB_SLAM2 抄送：wenhaoneu ,Author 刚刚加载完图片进入TRACK 就出错，这样子信息太少，看不出是哪里出错。

WeiYongqiang55, 2019-08-20:
我也于遇到了同样的问题，并不是depth下的图片不匹配的问题，我尝试过了，但是并没有解决问题

Blissy-32, 2024-04-08:
请问各位解决了么 我用的是自己单目深度估计生成的depth图片。输入后得到这样的结果