raulmur/ORB_SLAM2 #431: OpenCV Error:

Gasethata, 2017-09-13:


YuehChuan, 2017-09-17:
@Gasethata 
Same Issue! What's is your envionment
I'm using raspberry pi3 with ubuntuMATE16.04.3, ROS kinetic, and with OpenCV-3.3.0(newest one) installed
I guess is OpenCV version problem. ORB_SLAM2 just support toup to OpenCV-3.2.0

YuehChuan, 2017-09-17:
I;m using ros package usb_cam and  run mono camera demo with command
`rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monocular/head_camera.yaml /camera/image_raw:=/usb_cam/image_raw`

YuehChuan, 2017-09-17:
relate to
https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/31

YuehChuan, 2017-09-17:
quite wired, i already delete opencv-3.3.0 from */usr/local/lib*  */usr/local/include* 
then compile  everything make sure it is configured with kinetics opencv-3.2.0 using find_package(OpenCV 3.2.0 REQUIRED) , also open  GUI for ubnutuMATE
still same issue 
DBoW2, ORB_SLAM2 ros, ORB_SLAM2
[image]
[image]
[image]

YuehChuan, 2017-09-19:
seems relate
https://answers.ros.org/question/111186/opencv-error-null-pointer-null-or-empty-buffer-in-cvopenfilestorage/

Gasethata, 2017-09-19:
I have managed to solve the problem.
The problem was the settings file that contained my camera calibration parameters.
Do check it.

But from that being fixed, new problem came.

After running this command, I get a black window stating that its "WAITING FOR IMAGES" 
I think its got something to do with ROS.

Can someone please help me out here.

Gasethata, 2017-09-20:
Hi everyone
I managed to solve the "WAITING FOR IMAGES"  issue.
This is what i did:
1. roscore
2. roslaunch usb_cam-test.launch (make sure that you are in the .launch file directory)
  - for my case, usb_cam is the topic I'm using.
3.  rosrun ORB_SLAM2 Mono /home/vip/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/vip/ORB_SLAM2/Examples/ROS/ORB_SLAM2/SettingsFiles/microsoftWebcam.yaml /camera/image_raw:=/usb_cam/image_raw

BBlackB, 2018-04-10:
Hi @Gasethata, I have the same problem("OpenCV Error").
could you please talk about the details that how you solve the "OpenCV Error" problem?
Thanks in advance.

BBlackB, 2018-04-10:
I got it.
It is wrong style of yaml file that cause the problem.
Thanks for all guys!

CharlieHe, 2018-09-21:
@BBlackB  Yeah, you are right! Thanks!

zivAstra, 2018-10-02:
Can you be more specific as to what fixes the yaml problem?
Thanks

braineniac, 2019-02-03:
If anybody else stumbles upon this problem, [this][1] is how you change the yaml file

[1]: https://github.com/raulmur/ORB_SLAM2/issues/332

heshanchuan, 2020-03-26:
I fixed same problem
I find there is a blank space before %YAML:1.0
So I delete the blank space.

heshanchuan, 2020-03-26:
I fixed same problem. It's all about the YAML grammar.
I find there is a blank space before %YAML:1.0
So I delete the blank space.