raulmur/ORB_SLAM2 #432: Issue while building ORB_SLAM2 libgdal.so

apaulworks, 2017-09-14:
The following error occurs after following the README tutorial for building ORB_SLAM2.
Commands:
/ORB_SLAM2$ sudo chmod +x build.sh
/ORB_SLAM2$ sudo ./build.sh 

Output:
[ 62%] Built target ORB_SLAM2
[ 65%] Linking CXX executable ../Examples/RGB-D/rgbd_tum
[ 68%] Linking CXX executable ../Examples/Stereo/stereo_kitti
[ 71%] Linking CXX executable ../Examples/Monocular/mono_euroc
[ 75%] Linking CXX executable ../Examples/Stereo/stereo_euroc
[ 78%] Linking CXX executable ../Examples/Monocular/mono_kitti
[ 81%] Linking CXX executable ../Examples/Monocular/mono_tum

**//usr/lib/libgdal.so.1: undefined reference to `sqlite3_column_table_name'**
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:145: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
make[2]: Target 'CMakeFiles/rgbd_tum.dir/build' not remade because of errors.
CMakeFiles/Makefile2:215: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2

What can I do to resolve this? Thanks!

apaulworks, 2017-09-14:
Solved it. libORB_SLAM2.so was dependent on an Anaconda version of libsqlite.so. Changed link to the x86_64-linux-gnu version.

closed 2017-09-14 by apaulworks

wendaoyuchen, 2017-11-28:
I have the same problem with you.But how should I do to change the link to x86_64-linux-gnu version?

apaulworks, 2017-11-28:
Try the following command to see what files libORB_SLAM2.so is dependent on: 
_ldd libORB_SLAM2.so_
Then there is a linux command link that allows you to create a symbolic link to another file, similar to a shortcut in windows. The command would look like this: _sudo ln -sf /path-to-the-x86_64-linux-gnu-version-of-libsqlite.so/ libsqlite.so_ 
See here for more info on symbolic links: https://unix.stackexchange.com/questions/151999/how-to-change-where-a-symlink-points