raulmur/ORB_SLAM2 #436: Pose of  each camera frame of a monocular camera

Morpheusai, 2017-09-28:
**Hi, i'm using a monocular camera.** I want to get trajectory of every frame. ORB_SLAM and ORB_SLAM2 don't support out pose of each frame, but only the keyFrame. Please tell me why? thanks a lot.

teduav, 2017-12-11:
Hi, 
   you can use this in ros_mono.cc
   cv::Mat Tcw= mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());

ichlubna, 2021-02-12:
Hello, thank you @tednav for the answer. I have tried adding simply:
`std::cerr << SLAM.TrackMonocular(im,tframe) << std::endl;`
to
`Examples/Monocular/mono_tum.cc`
but this returns only keyframe poses as the OP mentioned. Is there anything specific in _ros_mono.cc_ that I need to get the position for every frame?