raulmur/ORB_SLAM2 #439: Difference in trajectory generated in SLAM and Localization Mode

fawadahm, 2017-10-10:
I take a scan of a street using ORB SLAM2 in SLAM mode and save the trajectory and the map file it generates. Then I feed the map file to ORB SLAM2 and run it in localization using the same street scan as input and save the trajectory again.

When I calculated the difference in the trajectory of the two files, the result I get is an average of around 1.5 meters along all three axes.

Where is this error coming from? Stereo matching? Pose optimization or ORB feature extraction?

shridharkini6, 2018-01-23:
@fawadahm
can you please tell me how to load saved trajectory & map files in localization mode to ORB_SLAM2?

BW25, 2018-10-10:
@shridharkini6 
Try this version of ORBSLAM2
https://github.com/Alkaid-Benetnash/ORB_SLAM2