raulmur/ORB_SLAM2 #440: Why is there no linearizeOplus() method in class EdgeSim3ProjectXYZ?

highlightz, 2017-10-11:
I think g2o optimizer needs the Jacobian of residual with respect to Vertex in order to perform an optimization, but in file types_seven_dof_expmap.h, class EdgeSim3ProjectXYZ has no Jacobian definition. Can anyone understand this?

AlejandroSilvestri, 2017-10-12:
@highlightz ,

I'm not sure if I understood you correctly, but I try an answer.

EdgeSim3ProjectXYZ is a child class of templated BaseBinaryEdge class, which implements a virtual linerarizeOplus.  This method is inherited by EdgeSim3ProjectXYZ: there is a linerarizeOplus in EdgeSim3ProjectXYZ, an inherited one.

g2o calculates the Jacobian using linearizeOplus.

I quote g2o's manual, in 6.1: "It offers for free the calculation of the Jacobians, via an implementation of the linearizeOplus method."

Again, I'm not sure if this is what you ask.

highlightz, 2017-10-13:
@AlejandroSilvestri This is exactly what I need to know, and thanks for your explanation. So this is my understanding: since we do not provide a linearizeOplus method, no **analytic derivatives** are provided, instead, the optimizer uses the **automatic derivatives** from base class linearizeOplus. Am I correct?

AlejandroSilvestri, 2017-10-13:
@highlightz ,

That's correct!

g2o has many base template classes, as helpers for specialization programmers should do.

But it also has a lot of very common specializations already implemented.  EdgeSim3ProjectXYZ is one of them.  If I remember correctly, orb-slam2 doesn't implement any specialization, it only uses those already provided by g2o.

highlightz, 2017-10-14:
@AlejandroSilvestri Thanks again.