raulmur/ORB_SLAM2 #441: Processing your own sequences.

Gasethata, 2017-10-12:
Greetings
I have tried to run my own sequences with the following challenges:

1. The current frame window is now enlarged (figured out is the frame size from my camera .i.e [1920x1080].
2. The ORB_SLAM2 way too slow (I have no idea what could be the problem), although there are features detected and the map is built.
3. What needs to be in my sequence directory (images / timestamps ..) [Is this enough?]

Note:
The sequence is of a hand-held monocular camera......and the other one is from a drone with a down-facing camera.

**Any help regarding this will be much appreciated.**

AlejandroSilvestri, 2017-10-13:
@Gasethata ,

I don't know why the low speed, but I wonder how did you calibrate those cameras.  I believe both have wide angle lens, needing fisheye undistortion model, which orb-slam2 doesn't provide.

Orb-slam2 has many examples, some for a variety of datasets, reading image files from a directory.  Adopt the names of images files of the desire dataset.

Gasethata, 2017-10-17:
@AlejandroSilvestri 

All the cameras I have used so far (USB webcams) are calibrated with ROS.

My Webcams:
1. Microsoft Lifecam VX-800 Model 1407 (Which I believe its not wide angle lens)
2. SJCAM HD 1080P

Any suggestions on which cameras to use if USB webcams are not suitable to use?
Adopting the names of images files of the desired dataset (not clear!)
All my image files are named in this manner 100001.png 

But funny enough, using the same cameras with their calibration files, the ORB_SLAM2 with ROS for real-time operation works well.

AlejandroSilvestri, 2017-10-17:
@Gasethata ,

If they are webcams, I believe they haven't wide angle lens.  So far so good.

Try running the TUM dataset example.  Download TUM dataset and run mono_tum .

Then replace TUM dataset image files with yours.  Edit yaml file and paste your webcam calibration parameters.

Gasethata, 2017-10-18:
@AlejandroSilvestri 

Thats exactly how I have been doing it this whole time. TUM dataset works like a charm.
I edited the .yaml file with my webcam calibration file. It works well in ROS for real-time operation.
The issue is when I try and use my image files.

AlejandroSilvestri, 2017-10-18:
@Gasethata ,

Before completely dropping the calibration issue, by any chance, do your images in files have the same resolution than the streaming video you successfully used for real-time operation?  If they don't, calibration is needed.

Returning to your original questions:

1. Current frame window adopts the image size.  I modified my code to let the user reduce this window.

2. ORB-SLAM2 is slow, compared to what?  Many videos published on internet are not in real-time, so they look fast but they aren't.  Actual speed depends on your hardware, on your microprocessor speed.  May be your images files are compressed, but decompression doesn't take that much time compared with orb-slam2.

3. This is enough.  You are already processing  your images files.  Timestamps aren't used in slam operation.

Gasethata, 2017-10-19:
@AlejandroSilvestri 

The image resolution is the same (640 x 480) in my image files and in real-time operation.

The ORB_SLAM2 is slow with my image sequences compared to when using the TUM dataset (freiburg1_xyz) as per ORB_SLAM2 Monocular example.

I believe my hardware should be good enough:

   Memory   15.6 GiB
Processor    Intel® Core™ i7-6700 CPU @ 3.40GHz × 8 
  Graphics    Intel® HD Graphics 530 (Skylake GT2) 
   OS type    64-bit
         Disk    137.9GB

AlejandroSilvestri, 2017-10-19:
@Gasethata 
Your hardware is excellent for executing orb-slam2.

Your observation is important: your images sequence runs slower than TUM's, so I believe something relating to your dataset it slowing down orb-slam2, but I can't figure it out.

Gasethata, 2017-10-24:
@AlejandroSilvestri 

I changed the names of my image sequences to the similar format of that of TUM and KITTI.
From **100001.jpg to 1305031102.170001.jpg** where only the last four digits changes incrementally.

I honestly don't know what effect that has, but the speed improved very well.

closed 2017-10-27 by Gasethata