raulmur/ORB_SLAM2 #451: Error while making ORB_SLAM2

akashshar, 2017-10-26:
Can someone please guide me? Hello I am getting the following error while building
Thank you very much in advance.

akash@akash:~/src/ORB_SLAM2$ ./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/akash/src/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/akash/src/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/akash/src/ORB_SLAM2/build
[ 76%] Built target ORB_SLAM2
[ 80%] Building CXX object CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o
[ 84%] [ 88%] [ 92%] Building CXX object CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o
[ 96%] [100%] Building CXX object CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o
Building CXX object CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o
Building CXX object CMakeFiles/stereo_euroc.dir/Examples/Stereo/stereo_euroc.cc.o
Building CXX object CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o
/home/akash/src/ORB_SLAM2/Examples/Monocular/mono_euroc.cc: In function ‘int main(int, char**)’:
/home/akash/src/ORB_SLAM2/Examples/Monocular/mono_euroc.cc:73:48: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
         im = cv::imread(vstrImageFilenames[ni],CV_LOAD_IMAGE_UNCHANGED);
                                                ^
/home/akash/src/ORB_SLAM2/Examples/RGB-D/rgbd_tum.cc: In function ‘int main(int, char**)’:
/home/akash/src/ORB_SLAM2/Examples/RGB-D/rgbd_tum.cc:80:74: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
         imRGB = cv::imread(string(argv[3])+"/"+vstrImageFilenamesRGB[ni],CV_LOAD_IMAGE_UNCHANGED);
                                                                          ^
/home/akash/src/ORB_SLAM2/Examples/Monocular/mono_kitti.cc: In function ‘int main(int, char**)’:
/home/akash/src/ORB_SLAM2/Examples/Monocular/mono_kitti.cc:68:48: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
         im = cv::imread(vstrImageFilenames[ni],CV_LOAD_IMAGE_UNCHANGED);
                                                ^
/home/akash/src/ORB_SLAM2/Examples/Stereo/stereo_euroc.cc: In function ‘int main(int, char**)’:
/home/akash/src/ORB_SLAM2/Examples/Stereo/stereo_euroc.cc:119:47: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
         imLeft = cv::imread(vstrImageLeft[ni],CV_LOAD_IMAGE_UNCHANGED);
                                               ^
/home/akash/src/ORB_SLAM2/Examples/Stereo/stereo_kitti.cc: In function ‘int main(int, char**)’:
/home/akash/src/ORB_SLAM2/Examples/Stereo/stereo_kitti.cc:69:47: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
         imLeft = cv::imread(vstrImageLeft[ni],CV_LOAD_IMAGE_UNCHANGED);
                                               ^
/home/akash/src/ORB_SLAM2/Examples/Monocular/mono_tum.cc: In function ‘int main(int, char**)’:
/home/akash/src/ORB_SLAM2/Examples/Monocular/mono_tum.cc:68:68: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
         im = cv::imread(string(argv[3])+"/"+vstrImageFilenames[ni],CV_LOAD_IMAGE_UNCHANGED);
                                                                    ^
make[2]: *** [CMakeFiles/mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o] Error 1
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o] Error 1
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
make[2]: *** [CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o] Error 1
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
make[2]: *** [CMakeFiles/stereo_euroc.dir/Examples/Stereo/stereo_euroc.cc.o] Error 1
make[1]: *** [CMakeFiles/stereo_euroc.dir/all] Error 2
make[2]: *** [CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o] Error 1
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2
make[2]: *** [CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o] Error 1
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
make: *** [all] Error 2

akashshar, 2017-10-26:
I got the solution, CV_LOAD_IMAGE_UNCHANGED is from the outdated since openCV 3.x.x
So, i simple replaced CV_LOAD_IMAGE_UNCHANGED with it's enum value -1

closed 2017-11-03 by akashshar

hWesleyWang, 2019-01-04:
You could add below in front of the cc file.

EX: in /xxxxxx/ORB_SLAM2/Examples/Monocular/mono_kitti.cc

#include<System.h>
#include <unistd.h>
**#define CV_LOAD_IMAGE_UNCHANGED -1  //new added one**

Shivani1796, 2019-09-06:
error: ‘CV_LOAD_IMAGE_GRAYSCALE’ was not declared in this scope
  cv::Mat m = cv::imread(filename, CV_LOAD_IMAGE_GRAYSCALE);
How to resolve this error ?? Can anybody help

jasonyi70, 2020-09-20:
Have you solve "Fatal error: opencv/cv.h: No such file or directory #include <opencv/cv.h> #782"?

hisan, 2021-10-30:
I solved this problem.
I meet the problem with use the opencv3.  so the method just for ORB_SLAM2 with opencv3.
(1) when execute the ./build.sh ,the error "/error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope " 
(2) so then I execute the cmd in terminal to find all file with the marco "CV_LOAD_IMAGE_UNCHANGED"
`find -type f -name "*.cc" -or -name "*.h" | xargs -i grep -l "CV_LOAD_IMAGE_UNCHANGED" {} `
(3) then I modifed all file find by step (2),that is add "#include <opencv2/imgcodecs/imgcodecs_c.h>" to these files.
(4) then re-exec the build.sh is ok!