raulmur/ORB_SLAM2 #453: Error while building ORB-SLAM2

zlingBryan, 2017-10-27:
Can someone please guide me? Hello I am getting the following error while building

/home/bryan/ORB_SLAM2/include/MapDrawer.h:43:28: error: ‘pangolin’ has not been declared
     void DrawCurrentCamera(pangolin::OpenGlMatrix &Twc);
                            ^
/home/bryan/ORB_SLAM2/include/MapDrawer.h:43:51: error: expected ‘,’ or ‘...’ before ‘&’ token
     void DrawCurrentCamera(pangolin::OpenGlMatrix &Twc);
                                                   ^
/home/bryan/ORB_SLAM2/include/MapDrawer.h:46:39: error: ‘pangolin’ has not been declared
     void GetCurrentOpenGLCameraMatrix(pangolin::OpenGlMatrix &M);
                                       ^
/home/bryan/ORB_SLAM2/include/MapDrawer.h:46:62: error: expected ‘,’ or ‘...’ before ‘&’ token
     void GetCurrentOpenGLCameraMatrix(pangolin::OpenGlMatrix &M);
                                                              ^
/home/bryan/ORB_SLAM2/src/Viewer.cc: In member function ‘void ORB_SLAM2::Viewer::Run()’:
/home/bryan/ORB_SLAM2/src/Viewer.cc:63:5: error: ‘pangolin’ has not been declared
     pangolin::CreateWindowAndBind("ORB-SLAM2: Map Viewer",1024,768);
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:66:14: error: ‘GL_DEPTH_TEST’ was not declared in this scope
     glEnable(GL_DEPTH_TEST);
              ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:66:27: error: ‘glEnable’ was not declared in this scope
     glEnable(GL_DEPTH_TEST);
                           ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:69:15: error: ‘GL_BLEND’ was not declared in this scope
     glEnable (GL_BLEND);
               ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:70:18: error: ‘GL_SRC_ALPHA’ was not declared in this scope
     glBlendFunc (GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA);
                  ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:70:32: error: ‘GL_ONE_MINUS_SRC_ALPHA’ was not declared in this scope
     glBlendFunc (GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA);
                                ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:70:54: error: ‘glBlendFunc’ was not declared in this scope
     glBlendFunc (GL_SRC_ALPHA, GL_ONE_MINUS_SRC_ALPHA);
                                                      ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:72:5: error: ‘pangolin’ has not been declared
     pangolin::CreatePanel("menu").SetBounds(0.0,1.0,0.0,pangolin::Attach::Pix(175));
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:72:57: error: ‘pangolin’ has not been declared
     pangolin::CreatePanel("menu").SetBounds(0.0,1.0,0.0,pangolin::Attach::Pix(175));
                                                         ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:73:5: error: ‘pangolin’ has not been declared
     pangolin::Var<bool> menuFollowCamera("menu.Follow Camera",true,true);
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:73:19: error: expected primary-expression before ‘bool’
     pangolin::Var<bool> menuFollowCamera("menu.Follow Camera",true,true);
                   ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:74:5: error: ‘pangolin’ has not been declared
     pangolin::Var<bool> menuShowPoints("menu.Show Points",true,true);
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:74:19: error: expected primary-expression before ‘bool’
     pangolin::Var<bool> menuShowPoints("menu.Show Points",true,true);
                   ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:75:5: error: ‘pangolin’ has not been declared
     pangolin::Var<bool> menuShowKeyFrames("menu.Show KeyFrames",true,true);
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:75:19: error: expected primary-expression before ‘bool’
     pangolin::Var<bool> menuShowKeyFrames("menu.Show KeyFrames",true,true);
                   ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:76:5: error: ‘pangolin’ has not been declared
     pangolin::Var<bool> menuShowGraph("menu.Show Graph",true,true);
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:76:19: error: expected primary-expression before ‘bool’
     pangolin::Var<bool> menuShowGraph("menu.Show Graph",true,true);
                   ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:77:5: error: ‘pangolin’ has not been declared
     pangolin::Var<bool> menuLocalizationMode("menu.Localization Mode",false,true);
     ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:77:19: error: expected primary-expression before ‘bool’
[... 67 lines cut ...]
                                 ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:136:12: error: ‘menuShowPoints’ was not declared in this scope
         if(menuShowPoints)
            ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:139:9: error: ‘pangolin’ has not been declared
         pangolin::FinishFrame();
         ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:145:12: error: ‘menuReset’ was not declared in this scope
         if(menuReset)
            ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:147:13: error: ‘menuShowGraph’ was not declared in this scope
             menuShowGraph = true;
             ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:148:13: error: ‘menuShowKeyFrames’ was not declared in this scope
             menuShowKeyFrames = true;
             ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:149:13: error: ‘menuShowPoints’ was not declared in this scope
             menuShowPoints = true;
             ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:150:13: error: ‘menuLocalizationMode’ was not declared in this scope
             menuLocalizationMode = false;
             ^
/home/bryan/ORB_SLAM2/src/Viewer.cc:155:13: error: ‘menuFollowCamera’ was not declared in this scope
             menuFollowCamera = true;
             ^

xdotzv, 2017-10-27:
Did you include the pangolin headers and link with pangolin lib?

zlingBryan, 2017-10-27:
@xdotzv   I do not. How can i do it? Thank you!

SteveMacenski, 2017-10-27:
Please search for simple errors yourself, these aren't github issues they're dependency problems. Google pangolin headers and I suspect you can handle the rest yourself.