raulmur/ORB_SLAM2 #454: How to evalute ORB-SLAM2 (and LSD-SLAM) on the TUM RGB-D Benchmark dataset?

haidela, 2017-10-27:
Good evening, How to evalute Absolute Trajectory Error(ATE) using KITTI dataset or EuRoC on Matlab?

MichaelGrupp, 2017-10-30:
This looks like multiple questions in one and your comment is unrelated to the header of your question, but I'll try to help. 

If you need to use MATLAB, you can for sure implement the ATE in MATLAB - maybe someone already did that (search). Since you want to use TUM RGB-D, KITTI and EuRoC datasets keep in mind that the "default" ground truth trajectory file formats of these datasets are different:
* TUM: timestamp + xyz + quaternion per row
* KITTI: flattened SE(3) pose matrix per row (without bottom row 0 0 0 1)
* EuRoC: csv with state information in IMU/body frame

Besides the *core* algorithm, you probably also want to align the trajectories to the ground truth - for this you can implement e.g. Horn or Umeyama alignment. Umeyama has the advantage that you can also get a scale correction which is useful for monocular SLAM like LSD-SLAM. See also my answer in issue #121.

I don't use MATLAB, but I implemented an evaluation package in Python that supports all three datasets and can do all kinds of alignment: [evo](https://github.com/MichaelGrupp/evo). Maybe it's useful for you.

haidela, 2017-10-30:
thank you very much, it works well

closed 2017-11-03 by haidela