raulmur/ORB_SLAM2 #459: Camera is pointing at the ground, so robot is climbing?

bmende, 2017-11-03:
On my wheeled robot, the camera is pointing slightly at the ground, so ORB_SLAM2 seems to think that the robot is climbing (since the ground is sloped up in the camera reference frame). It comes back 'down' when going in a loop, but is there a way to get ORB_SLAM to re-orient with the ground plane down? What if I can find the orientation of the camera by hand? where would I pass in the correction?

sunkaianna, 2017-11-03:
if your robot is always moving on a “plane”,i suggest you can just update 3dof poses.

AlejandroSilvestri, 2017-11-13:
@bmende ,

ORB-SLAM2's world reference is arbitrary.  It uses Initial Frame as origin (position and orientation), and then it takes scale from median depth in initial map.

If you need a different reference, let say a real world reference, you should calculate your own 4x4 transformation matrix, and apply it to frames poses.