raulmur/ORB_SLAM2 #462: Doubt about the range of disparity： maxD

NovenBae, 2017-11-06:
In the function of the Frame.cpp, I have some doubts about the const variables: 

const float minZ = mb; 
const float minD = 0;
const float maxD = mbf / minZ;

why the value of minZ is not fx but mb? and why the value of maxD is not mb?

AlejandroSilvestri, 2017-11-12:
@NovenBae 

You mean [here](https://github.com/raulmur/ORB_SLAM2/blob/master/src/Frame.cc#L495).  And from [here](https://github.com/raulmur/ORB_SLAM2/blob/master/include/Frame.h#L121):

    // Stereo baseline multiplied by fx.
    float mbf;

    // Stereo baseline in meters.
    float mb;

So mbf = mb * fx  and maxD = fx.

mb is n meters, fx is in pixels.

minZ is only used in those lines, only to clarify the reasoning.  They could have used fx instead of maxD, and 0 instead of minD.

minZ is defined in meters, arbitrarily equal to mb, forming a sort of square mb x minZ.  In that square, at a minZ distance from camera the deviation maxD in pixels will be equal to fx.

takuhara, 2018-01-15:
maxD = fx.
so We will search from the disparity search range of stereo matching from -3 to around 400.
A very close 3D point was mistakenly accidentally generated and failed.

I made it like "const float maxD = 30;", I went very well.
Should I limit maxD?