raulmur/ORB_SLAM2 #463: virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()

haidela, 2017-11-06:
Hello everyone, I am wanting to test Running Stereo Node example, after running, I found this error: virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
I pressed Ctrl + C for saving the trajectory, Would it not have consequences on the trajectories achieved ?
Thank you in advance for your answer.

1. haidara@haidara-virtual-machine:~$ roscore
... logging to /home/haidara/.ros/log/ba6bf0a2-c321-11e7-84f4-000c29ce0138/roslaunch-haidara-virtual-machine-2893.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://haidara-virtual-machine:45571/
ros_comm version 1.11.21
SUMMARY
========
PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
NODES
auto-starting new master
process[master]: started with pid [2905]
ROS_MASTER_URI=http://haidara-virtual-machine:11311/

setting /run_id to ba6bf0a2-c321-11e7-84f4-000c29ce0138
process[rosout-1]: started with pid [2918]
started core service [/rosout]

2. haidara@haidara-virtual-machine:~/catkin_ws/src/ORB_SLAM2$ rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
Input sensor was set to: Stereo
Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!
Camera Parameters: 
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)
ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
Depth Threshold (Close/Far Points): 3.85272
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.New map created with 542 points
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 766 points
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 794 points
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 608 points
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 731 points
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()

3. haidara@haidara-virtual-machine:~/catkin_ws/src/ORB_SLAM2/Examples/Stereo$ rosbag play --pause V1_01_easy.bag /cam0/image_raw:=/camera/left/image_raw /cam1/image_raw:=/camera/right/image_raw
[ INFO] [1509996062.846894567]: Opening V1_01_easy.bag
Waiting 0.2 seconds after advertising topics... done.
Hit space to toggle paused, or 's' to step.
 [PAUSED]   Bag Time: 1403715271.716820   Duration: 0.000000 / 147.166583 [PAUSED]   Bag Time: 1403715271.716820   Duration: 0.000000 / 147.166583 [PAUSED]   Bag Time: 1403715271.716820   Duration: 0.000000 / 147.166583 [PAUSED]   Bag Time: 1403715271.716820   Duration: 0.000000 / 147.166583 [PAUSED]   Bag Time: 1403715271.716820   Duration: 0.000000 / 147.166583
[RUNNING]  Bag Time: 1403715417.256565   Duration: 145.539745 / 147.1665 [RUNNING]  Bag Time: 1403715417.258529   Duration: 145.541710 / 147.1665 [RUNNING]  Bag Time: 1403715418.863686   Duration: 147.146866 / 147.1665 [RUNNING]  Bag Time: 1403715418.870562   Duration: 147.153743 / 147.1665 [RUNNING]  Bag Time: 1403715418.873686   Duration: 147.156866 / 147.166583     
Done.

LongruiDong, 2020-06-04:
Hi,
I also met similiar bug, do you know where the problem is?

yanyan-li, 2020-06-04:
Hi, in my code.  sometimes, if points are regarded as outliers in trackmotionmodel(), there are not enough points in tracklocalmap()'s poseoptimization() function.