raulmur/ORB_SLAM2 #467: Interpreting the KeyFrameTrajectory.txt file

Gasethata, 2017-11-10:
Hi everyone

I used a drone dataset from NPU Drone-Map Dataset and it has GPS trajectory of the flight.
After using ORB-SLAM2 on this dataset, I would like to plot the GPS data together with the KeyFrameTrajectory.txt data to see if they correspond.

My challenge now is that I dont know how to interpret data from this KeyFrameTrajectory.txt file. I'm not that sure myself that what I wanna do is doable.

I need like (x,y,z) co-odinates from this file.

Can someone please help me on this. 

below is a sample of what is in my KeyFrameTrajectory.txt file.....and I don't know what is what.
Thank you.

1305031102.170000 0.0365327 -1.4422618 -0.0842586 -0.0011281 0.0000323 0.0284874 0.9995935
1305031102.170000 0.0345594 -1.5336652 -0.0918721 0.0012357 0.0007751 0.0287673 0.9995851
1305031102.170000 0.0341972 -1.5458909 -0.0928111 0.0015669 0.0010044 0.0287377 0.9995853
1305031102.170000 0.0304142 -1.6184912 -0.0992637 0.0033218 0.0024995 0.0287309 0.9995785
1305031102.170000 0.0253411 -1.6931344 -0.1061403 0.0038579 0.0046472 0.0283241 0.9995806
1305031102.170000 0.0245110 -1.7027667 -0.1081518 0.0052676 0.0050030 0.0282190 0.9995754

akashshar, 2017-11-10:
Did you get any solution for this? I am also stuck at this point

MichaelGrupp, 2017-11-11:
This is the trajectory (timestamps + 3D poses) that was estimated by the SLAM algorithm. More specifically, it has only the poses of the keyframes. It's saved in the file format that was introduced by the TUM RGB-D dataset ("TUM format", see https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats).

Every row has 8 entries containing timestamp (in seconds), position and orientation (as quaternion):
`timestamp x y z q_x q_y q_z q_w`

So to achieve what you want to do, you could for example load the file as a table (similar to a .csv file) and then the columns 2 to 4 are your x, y, z values (or 1 to 3 if you count from 0).

---

Maybe also useful for you: I made a Python package for SLAM trajectories (see [here](https://www.github.com/MichaelGrupp/evo)), it has executables that can parse files in the TUM format - e.g. you could inspect and plot your file with the command:
```
evo_traj tum KeyframeTrajectory.txt --plot
```

akashshar, 2017-11-14:
@MichaelGrupp Thank you very much!

NeXT1995, 2018-06-13:
@akashshar @Gasethata I also tried to combine GPS data with KeyFrameTrajectory.txt data，Have you solved this problem?

Gasethata, 2018-06-13:
@NeXT1995 
I plotted each of them separately...which of-cause can be combined plane. I used matlab for the plotting.
For KeyFrameTrajectory data (x,y,z) co-ordinates are the first three columns in the file (excluding the names of the image in the first column
[image]
).
And for GPS data, I again used matlab to extract only the (x,y,z) co-ordinates in each line, then store those in a file to use to plot.

rahulsharma11, 2018-08-03:
Hi, i am doing the same by running orbslam on my jetson tx2. The script is awesome. But  the thing is it works only with the stored file. Is it not possible to have some sort of socket programming so that x,y,z can be plot realtime on my desktop through socket? Is it possible?

WWBlueFire, 2018-11-29:
Thanks for you sharing, when I input "evo_traj tum KeyframeTrajectory.txt --plot" in the command line in Ubuntu16.04, unexpected error occurred as blow:
name:	KeyFrameTrajectory18112903
infos:	86 poses, 1.429m path length, 217.468s duration
Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/lib/python3.5/tkinter/__init__.py", line 36, in <module>
    import _tkinter
ImportError: No module named '_tkinter'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zz/.local/lib/python3.5/site-packages/evo/entry_points.py", line 89, in launch
    main_module.run(args)
  File "/home/zz/.local/lib/python3.5/site-packages/evo/main_traj.py", line 339, in run
    from evo.tools.plot import PlotMode
  File "/home/zz/.local/lib/python3.5/site-packages/evo/tools/plot.py", line 35, in <module>
    import matplotlib.pyplot as plt
  File "/home/zz/.local/lib/python3.5/site-packages/matplotlib/pyplot.py", line 2374, in <module>
    switch_backend(rcParams["backend"])
  File "/home/zz/.local/lib/python3.5/site-packages/matplotlib/pyplot.py", line 207, in switch_backend
    backend_mod = importlib.import_module(backend_name)
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/zz/.local/lib/python3.5/site-packages/matplotlib/backends/backend_tkagg.py", line 1, in <module>
    from . import _backend_tk
  File "/home/zz/.local/lib/python3.5/site-packages/matplotlib/backends/_backend_tk.py", line 5, in <module>
    import tkinter as Tk
  File "/usr/lib/python3.5/tkinter/__init__.py", line 38, in <module>
    raise ImportError(str(msg) + ', please install the python3-tk package')
ImportError: No module named '_tkinter', please install the python3-tk package

evo module evo.main_traj crashed - no logfile written (disabled)
////////////////////////////////////////////////////////////////////////////////////////
I spent a lot of time solving the problem,but now I can't solve it by myself. I wish you answer,if you know.
thanks a lot in advance!!

MichaelGrupp, 2018-11-29:
```
ImportError: No module named '_tkinter', please install the python3-tk package
```
It's already saying what you need to do. Try something like this: https://askubuntu.com/a/950707

MichaelGrupp, 2018-11-29:
> Hi, i am doing the same by running orbslam on my jetson tx2. The script is awesome. But the thing is it works only with the stored file. Is it not possible to have some sort of socket programming so that x,y,z can be plot realtime on my desktop through socket? Is it possible?

I'd use ROS and RViz or something like rqt_plot for real-time visualization.

immuhammadadil, 2019-04-06:
Can we bypass SLAM to produce trajectory file with timestamp and other 7 parameters if we have GPS information, Altitude of Quadcopter and also roll pitch & yaw??

kuntalgh24, 2020-05-22:
Hello Michael 
whenever I am trying to plot the trajectories I am facing following error which is leaving me in perplexed situation. Please help accordingly.
Thanks in Advance.
Kuntal
name:	s2-KeyFrameTrajectory
infos:	296 poses, 6.849m path length, 147.197s duration
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/kuntal/.local/lib/python3.5/site-packages/evo/entry_points.py", line 92, in launch
    main_module.run(args)
  File "/home/kuntal/.local/lib/python3.5/site-packages/evo/main_traj.py", line 464, in run
    plot_collection.show()
  File "/home/kuntal/.local/lib/python3.5/site-packages/evo/tools/plot.py", line 194, in show
    self.tabbed_tk_window()
  File "/home/kuntal/.local/lib/python3.5/site-packages/evo/tools/plot.py", line 151, in tabbed_tk_window
    from matplotlib.backends.backend_tkagg import FigureCanvasTkAgg, NavigationToolbar2Tk
ImportError: cannot import name 'NavigationToolbar2Tk'

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

MichaelGrupp, 2020-05-23:
You need to update matplotlib to a newer version

arunceria, 2020-06-23:
Good day.
How can I check the distance in meters from the keyframe to the real one?

Thanks

KARAASLAN-AI, 2022-01-07:
> Can we bypass SLAM to produce trajectory file with timestamp and other 7 parameters if we have GPS information, Altitude of Quadcopter and also roll pitch & yaw??

were you able to do this?

NeXT1995, 2022-01-07:
您好，你的邮件，杨沂林已经收到，我会尽快查收的，谢谢！

TRINHHOANGANH, 2023-02-22:
Hi @All
Can i use KeyframeTrajectory.txt  data to add coordinates of previous keyframes to next image such as image below
[image]
Thanks