raulmur/ORB_SLAM2 #472: Initialization issue with ZED camera

InigoCastro, 2017-11-14:
Hi all, 

I was trying to make use of an image input generated with the ZED camera with ORB_SLAM2 . For this, I calibrated both cameras separately and also as a stereo system to get the extrinsic parameters. This allowed me to create the necessary documents for running ORB_SLAM  (I mimicked the inputs that the EuRoC data set provides, and tailored the setting files for the ZED camera). For running ORB_SLAM2 I also use the file _stereo_euroc.cpp_ However, after going through this process and trying to run the code, the Initialization seems to be failing. Only the first frame shows in the  ´Current frame´ window and never changes. Below the frame, the message ´TRYING TO INITIALIZE´ is displayed. The terminal output for this situation is the following:

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 1617.51
- fy: 1617.51
- cx: 972.852
- cy: 540.584
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 15
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 5.25245

-------
Start processing sequence ...
Images in the sequence: 36

New map created with 366 points
init done
opengl support available

I was wondering if you had any experience with this and could help me debug the root of this problem, as I have tried several things but nothing has worked so far.

sunkaianna, 2017-11-14:
Maybe your image sequence is TOO short to have enough motions. The outputs shows the orbslam2 system was working.

Pooja71, 2017-11-14:
Like @sunkaianna suggested, your recorded dataset might be short. Does the algorithm exit or is still running? Did you try giving live feed from the camera? Are you using ROS based version? I have worked on this before and maybe able to help.

InigoCastro, 2017-11-15:
A thing just happened that makes me think that the process may be running really slowly. I left it there for awhile and I came back to this: The first frame of this short sequence was showing (and static, or it may be that it was the second frame) but no warning prompted and the points were actually showing in the frame

I just tried with a sequence of 430 images and it still shows the warning. The algorithm keeps working in the sense that ORB_SLAM2 does not close on itself, but it seem that it never manages to initialize.

No, I have not yet tried to use the live feed from the camera as my intention was to mimic the EuRoC data set, by running a certain sequence with ORB_SLAM2. My idea is, as soon as I am able to get it to work, to start looking into the ROS-wrapper of ZED to make it work with ORB_SLAM2 with live feed. 

I am currently using the opensource ORB_SLAM2 code and a self-made script to capture the sequence images from ZED at 15 fps and HD720p that are used as input for ORB_SLAM2.

InigoCastro, 2017-11-21:
The problem still persists. I reduced the resolution from the ZED camera output to VGA (15 fps) and calibrated it again. Tweaked the input files for OBR_SLAM2 and still shows ´TRYING TO INITIALIZE´ in the current frame view while being stuck in the same frame. The image sequence comprises 448 images

Does anyone have a clue about why this could be happening? Could any of the input parameters of the file similar to _EuRoC.yaml_ that I generated be affecting the process? I have yet to correctly determine **Camera.bf** (currently using **42** for VGA) and **ThDepth** (**50**).
May I need to tweak the ORB parameters (which can also be found in the aforementioned file) to make it work?

Help on this would be much appreciated, as I am running out of ideas.

InigoCastro, 2017-11-21:
I fixed it. The problem is that since I did not know what time units to use for the time stamps, I was trying to get to the same units as int he _EuRoC_ data set (by looking a the _data.csv_ files). For this, I was adding a certain amount of zeros as "000" string at the end of the image filenames (  "345000.png" instead of  "345,png"). It seems that this blocked ORB_SLAM2 from making proper use of the images. After I deleted these zeroes, it started working just fine.

closed 2017-11-21 by InigoCastro

HelenXR, 2018-01-21:
Looks great.
@InigoCastro 
I am a beginner, I also want to run ORB_SLAM2 with a ZED camera. Would  you mind to share your details steps or demo?

sivaariram, 2018-12-12:
@Pooja71 what is the command for live feed from zed stereo camera?