raulmur/ORB_SLAM2 #480: usage

kamel271, 2017-11-30:
I am new with SLAM so  could you please tell me what is meaning by the last sentence Usage: Usage: rosrun ORB_SLAM2 Mono path_to_vocabulary path_to_settings i get this when ever i run this command rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monoculer/EuRoC.yaml true
[image]

lfrschkn, 2017-11-30:
hi
this is a message showing you how to use the examples
it is telling you to structure your command as:
rosrun ORB_SLAM2 Mono path_to_vocabulary path_to_settings

you basically did that but I think you have got the path_to_settings wrong
you have: Examples/Monoculer/...
it should be: Examples/Monocular/...

kamel271, 2017-12-01:
thank you for the answer  lfrschkn

lfrschkn, 2017-12-03:
refer to [#186](https://github.com/raulmur/ORB_SLAM2/issues/186)

closed 2017-12-20 by kamel271