raulmur/ORB_SLAM2 #488: How can I augment a point

akashshar, 2017-12-08:
I have calculated a point in 3d point (based on my requirements) and I am able to display this 3d point in the 3d map viewer using  glVertex3f(x,y,z); and it is working fine.

But I also want to view this in the 2d video stream as a simple augmentation. How can I do this? I understand that i need to multiply the 3d point with the projection matrix but I'm not exactly knowing how to access the projection matrix in ORB SLAM2. Any help would be appreciated. Thanks!!

AlejandroSilvestri, 2017-12-11:
@akashshar 

mCurrentFrame.mTcw is the camera pose.

mTcw * point3D gives you the point's coordinates in the camera reference.

mCurrentFrame.mK is the camera matrix.  You can use it to project the point and get its coordinates in pixels.

Sorry for being so brief.

akashshar, 2017-12-19:
Hey. Thank you very much for the reply. This was useful :)