raulmur/ORB_SLAM2 #490: Libopencv_calib3d3 error

chaiein, 2017-12-13:
How to solve the following error? 
[  0%] Built target rosbuild_precompile
make[2]: *** No rule to make target '/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2.0', needed by '../Mono'.  Stop.
CMakeFiles/Makefile2:718: recipe for target 'CMakeFiles/Mono.dir/all' failed
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2.0', needed by '../RGBD'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make[2]: *** No rule to make target '/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2.0', needed by '../Stereo'.  Stop.
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
make[2]: *** No rule to make target '/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2.0', needed by '../MonoAR'.  Stop.
CMakeFiles/Makefile2:820: recipe for target 'CMakeFiles/MonoAR.dir/all' failed
make[1]: *** [CMakeFiles/MonoAR.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

indikabw, 2017-12-22:
This happened to me just now after a Ubuntu update, which apparently upgraded libopencv_calib3d3.so.3.2.0 to libopencv_calib3d3.so.3.3.1. Deleting the build and devel folders and doing a catkin_make fixed it in my case.

AldrichCabrera, 2017-12-27:
In my case was the same but I reinstall ROS and for bad luck, the new ROS already have libopencv_calib3d3.so.3.3.1. Then I Deleted the build and devel folders but nothing it solve and appear new errors like:

/usr/bin/ld: CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
/usr/lib/x86_64-linux-gnu/libboost_system.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/RGBD.dir/build.make:217: recipe for target '../RGBD' failed
make[2]: *** [../RGBD] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2

chaiein, 2018-01-09:
deleting devel and build didnot solve the problem for me. Still have same error
:-1: error: No rule to make target '/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2.0', needed by 'RGBD'.  Stop.

AldrichCabrera, 2018-01-09:
Try put -lboost_system in:
set(LIBS 
${OpenCV_LIBS} 
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM2.so
-lboost_system
)

Yvon-Shong, 2018-01-15:
`Example/ROS/CMakeLists.txt`
set(LIBS
${OpenCV_LIBS}
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM2.so
**-lboost_system**
)

randoms, 2018-01-15:
This is a bug of OpenCV 3.3.1 which is the default version of ROS kinetic OpenCV3.
You can find the issue [here](https://github.com/opencv/opencv/issues/10105) and [here](https://github.com/ros-perception/vision_opencv/issues/193)

Run the following commands can fix this issue.
```
sed -i -r \
  's#set\(OpenCV_LIBS \$\{OpenCV_LIBS\} "\$\{__cvcomponent\}"\)#set\(OpenCV_LIBS \$\{OpenCV_LIBS\} "\$\{OpenCV_INSTALL_PATH\}/lib/lib\$\{__cvcomponent\}3.so")#' \
  /opt/ros/kinetic/share/OpenCV-3.3.1/OpenCVConfig.cmake
```

raktimkashyap, 2018-03-20:
@Yvon-Shong
I am facing the same issue and have inserted the line  "-lboost_system" just like you  have suggested.. but it still shows the same  errors. Can you suggest some other way?

Regards

chrisspen, 2018-06-25:
>This happened to me just now after a Ubuntu update, which apparently upgraded libopencv_calib3d3.so.3.2.0 to libopencv_calib3d3.so.3.3.1. Deleting the build and devel folders and doing a catkin_make fixed it in my case. I'm working on a different package, but could still be useful.

That worked for me. Thanks.

salahkhan94, 2018-11-08:
@raktimkashyap  I'm facing the same issue were you able to resolve it?