raulmur/ORB_SLAM2 #493: “Segmentation fault (core dumped)”when run the g2o::EdgeSim3* setMeasurement()

ZulunZhu, 2017-12-27:
Hi
I met the Segmentation fault when I use the g2o imitating the ORB-SLAM2 to optimize my pose graph.

Eigen::Matrix<double,3,3> Rcw;
Eigen::Matrix<double,3,1> tcw;
Rcw<<0.999991,-0.00307419,-0.00307221,
0.00307749,0.999995,0.00106957,
0.00306891,-0.00107902,0.999995;
tcw<<-0.00634032,
-0.0152401,
0.849042;
g2o::Sim3 S_test(Rcw,tcw,1.0);
cout<<"S_test:"<<S_test<<endl;
g2o::EdgeSim3* e = new g2o::EdgeSim3();
cout << "e.nIDj " << nIDj << " e.nIDi " << nIDi << endl;
e->setVertex(1, dynamic_castg2o::OptimizableGraph::Vertex*(optimizer.vertex(nIDj)));
e->setVertex(0, dynamic_castg2o::OptimizableGraph::Vertex*(optimizer.vertex(nIDi)));
e->setMeasurement(S_test);

I tried a lot of value of setMeasurement(),but every time my code will go core dumping at this function.
It is very confusing that ORB-SLAM2 will not go down in that function.

Does anyone have the same problem? Appreciating if there is any help.

chaiein, 2017-12-27:
I think you should give proper path of eigen files for EIGEN directory variable  in cmake_gui

ZulunZhu, 2017-12-27:
@chaiein 
Thank you for your reply!
I found when I  add the "set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3 -march=native")” in the cmakelist my code can run normally .But I really don't know why .
The problem is that when I add this to the cmakelist the other code will go core dumping,so do you have any advice to run g2o without the "-Wall -O3"?

hmchung, 2018-05-14:
@ZulunZhu Thank you for your sharing! Works for me too! Amazing... How did you even think of this...?

alibenD, 2018-12-27:
@ZulunZhu Thank you for you answer, it solves the problem I met(On MacOS10.14.2).  I guess `-march=native` is the key. The problem has confused me very long time.

JunaidCS032, 2019-01-09:
@chaiein Thanks for the solution. Phew.. after a long time, finally I get the solution.

Thanks a lot.
PS: @ZulunZhu  Thanks for the pointer to @chaiein