raulmur/ORB_SLAM2 #496: Try to build point cloud following raulmur's paper

ouerkakaChango, 2017-12-29:
Hello,everyone.
In the paper
ORB-SLAM2: an Open-Source SLAM System for Monocular, Stereo and RGB-D Cameras
(second paper in readme)
In Fig1(b)，the paper says:
"The pointcloud is rendered by backprojecting the sensor depth maps from estimated keyframe pose".
When I use TUM dataset "rgbd_dataset_freiburg1_desk",try to reconstruct point cloud by using keyframe,however,the point cloud pieces(from every keyframe,transformed by ORM-SLAM2‘s output pose) just don't align.
1.Is the reason of wrong chosen of dataset?
2.Is the reson of inaccuracy of ORB-SLAM2's output?
3.Is the reson of my code or something misunderstanding?
Hope to know the reason,thanks.
Here's my main part code of reconstructing full point cloud:
              //The rotation quaternion(x,y,z,w)
              double x = keyVec[i].qx, y = keyVec[i].qy, z = keyVec[i].qz, w = keyVec[i].qw;
              cv::Mat R;
             //convert to  rotation matrix
              R = (cv::Mat_<double>(3, 3)<<
              2*(x*x+w*w)-1,2*(x*y+z*w),2*(x*z-y*w),
              2*(x*y-z*w), 2*(y*y+w*w)-1, 2*(y*z+x*w),
             2*(x*z+y*w), 2*(y*z-x*w), 2*(z*z+w*w)-1
              );
                //This looks a little redundancy,just ignore...but doesn't affect result
		Eigen::Matrix3d r;
		cv::cv2eigen(R, r);
		Eigen::Isometry3d T = Eigen::Isometry3d::Identity();
		Eigen::AngleAxisd angle(r);
		T = angle;
               //add the position transfom to the matrix T
		T(0, 3) = keyVec[i].tx;
		T(1, 3) = keyVec[i].ty;
		T(2, 3) = keyVec[i].tz;
		// pcVec is the array of every frame's pointCloud
                //for every i∈[1,frameMaxID]
               //make every i2∈[0,i-1] transfom pose,to align to frame[i]
		for (int i2 = 0; i2 < i; i2++) {
			PointCloud::Ptr toutput(new PointCloud());
			pcl::transformPointCloud(*pcVec[i2], *toutput, T.matrix());
			pcVec[i2] = toutput;
		}
	}

ouerkakaChango, 2017-12-29:
oh,I've change the dataset to fr1_room as the paper does.The result is still not good...

ouerkakaChango, 2017-12-29:
OK,I solved it...
There were some misunderstanding...
1.Every column in the output is absolute transform,not relative transform from the front.So in my
code I shouldn't do that i2 thing.
2.When you try to align the piece to the same coordinate,you can use my T matrix,
different from the T above,the rotation is inversed,transform still the same.
Here's the code I modified:
                R = R.inv();
		Eigen::Matrix3d r;
		cv::cv2eigen(R, r);
		Eigen::Isometry3d T = Eigen::Isometry3d::Identity();
		Eigen::AngleAxisd angle(r);
		cout << "translation" << endl;
		T = angle;
		T(0, 3) = keyVec[i].tx;
		T(1, 3) = keyVec[i].ty;
		T(2, 3) = keyVec[i].tz;
                PointCloud::Ptr toutput(new PointCloud());
		pcl::transformPointCloud(*pcVec[i], *toutput, T.matrix());
		pcVec[i] = toutput;

The final dense point cloud is very accurate and pretty,thank you raulmur!...

closed 2017-12-29 by ouerkakaChango

YomnaJehad, 2019-04-21:
Do you possibly still have the full code after editing? I'm a total newbie trying to publish a point cloud

omair-khalid, 2019-07-11:
It would be great if you can share the code for this @ouerkakaChango

ouerkakaChango, 2019-07-12:
> 
> 
> It would be great if you can share the code for this @ouerkakaChango

I‘m sorry guys,the code was written on the computer in the company that I've left.It may still exist in the world,but I don't have it.

ouerkakaChango, 2019-07-12:
There were more part of the code I shared in my Chinese blog.
https://blog.csdn.net/qq_41524721
But for sure I wouldn't try to 'rebuilt' the code any more.