raulmur/ORB_SLAM2 #498: Can't build nodes for mono, stereo, RGB-D

eraserpencil, 2018-01-03:
Following the steps included in https://github.com/Abhimovel/mov/tree/master/ORB_SLAM2 Section #7. ROS Examples.

At the instruction "make",
I get this 
`/usr/bin/ld: CMakeFiles/Stereo.dir/src/ros_stereo.cc.o: undefined reference to symbol '_ZN5boost7archive17archive_exceptionD2Ev'
//usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.58.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/Stereo.dir/build.make:221: recipe for target '../Stereo' failed
make[2]: *** [../Stereo] Error 1
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
`

Referencing possible solutions from https://github.com/raulmur/ORB_SLAM2/issues/494, my new CMakeLists.txt is now 

`
set(LIBS 
${OpenCV_LIBS} 
${EIGEN3_LIBS} 
${Pangolin_LIBRARIES} 
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so 
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so 
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM2.so 
++
${PROJECT_SOURCE_DIR}/../../../lib/libboost_filesystem.so 
${PROJECT_SOURCE_DIR}/../../../lib/libboost_system.so
-lboost_system
`

and now the error 

`/usr/bin/ld: cannot find -libboost_system
collect2: error: ld returned 1 exit status
CMakeFiles/Stereo.dir/build.make:221: recipe for target '../Stereo' failed
make[2]: *** [../Stereo] Error 1
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
`

AldrichCabrera, 2018-01-09:
Could send the text of your terminal?

eraserpencil, 2018-01-09:
PBJ@SW-MBP:~/ORB_SLAM2$ rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true
[rosrun] Couldn't find executable named Stereo below /home/PBJ/ORB_SLAM2/Examples/ROS/ORB_SLAM2

Have successfully ran ./build.sh, there was no build_ros.sh executable, have exported ROS_PACKAGE_PATH, have changed the datasets as advised.

AldrichCabrera, 2018-01-09:
I assume like you ROS_PACKAGE_PATH is: 
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/PBJ/ORB_SLAM2/Examples/ROS

This line should be inside the file ~/.bashrc in my case the ROS_PACKAGE is inside the ROS workspace like:
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/user89/catkin_ws/src/ORB_SLAM2/Examples/ROS

Try to do a source the file .bashrc: source ~/.bashrc, then delete the build folder inside ROS/ORB_SLAM2 and launch again ./build_ros.sh

cyberphantom, 2018-01-16:
Hi I'm trying to run ORB_SLAM2 Mon, so I'm having this error when I run ./build_ros
I already included the export library and source ~/.bashrc file, any hint please?

mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM2
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using CATKIN_DEVEL_PREFIX: /home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel
-- Using CMAKE_PREFIX_PATH: /home/saif/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/saif/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- Using these message generators: gencpp;genlisp;genpy
[rosbuild] Including /opt/ros/indigo/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/indigo/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/indigo/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++11.
CMake Error at CMakeLists.txt:42 (find_package):
  By not providing "FindPangolin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Pangolin",
  but CMake did not find one.

  Could not find a package configuration file provided by "Pangolin" with any
  of the following names:

    PangolinConfig.cmake
    pangolin-config.cmake

  Add the installation prefix of "Pangolin" to CMAKE_PREFIX_PATH or set
  "Pangolin_DIR" to a directory containing one of the above files.  If
  "Pangolin" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
See also "/home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
saif@saif-Studio:~/catkin_ws/src/ORB_SLAM2$ clear

saif@saif-Studio:~/catkin_ws/src/ORB_SLAM2$ ./build_ros.sh
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM2
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using CATKIN_DEVEL_PREFIX: /home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel
-- Using CMAKE_PREFIX_PATH: /home/saif/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/saif/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- Using these message generators: gencpp;genlisp;genpy
[rosbuild] Including /opt/ros/indigo/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/indigo/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/indigo/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++11.
CMake Error at CMakeLists.txt:42 (find_package):
  By not providing "FindPangolin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Pangolin",
  but CMake did not find one.

  Could not find a package configuration file provided by "Pangolin" with any
  of the following names:

    PangolinConfig.cmake
    pangolin-config.cmake

  Add the installation prefix of "Pangolin" to CMAKE_PREFIX_PATH or set
  "Pangolin_DIR" to a directory containing one of the above files.  If
  "Pangolin" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
See also "/home/saif/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.

cdlwhm1217096231, 2018-01-26:
you should build your Examples/ROS/ORB_SLAM2,then rosrun package

liu-felix123, 2022-02-21:
> I assume like you ROS_PACKAGE_PATH is: export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/PBJ/ORB_SLAM2/Examples/ROS
>
> This line should be inside the file ~/.bashrc in my case the ROS_PACKAGE is [...]

hi, i followed your mehtod to set, but it wasn't work. could you give me another advise to solve the problem?  the  text of mine terminal as below:$ rosrun ORB_SLAM2 RGBD /home/lfh/catkin_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/lfh/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/Asus.yaml
[rosrun] Couldn't find executable named RGBD below /home/lfh/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2

look forward you reply,sincerely.