raulmur/ORB_SLAM2 #499: The order of XYZ in Twc/KeyFrameTrajectory?

EchizenG, 2018-01-03:
I have explored many issues here and find the order you offered in KeyFrameTrajectory.txt was "timestamp,x,y,z,qw,qx,qy,qz". But as far as I mesured, it is "timestamp,y,z,x,...". I also outputed the current frame's position "Twc". And it is "y,z,x" too.

So I came here to get your confirmation.

Thank you!

ouerkakaChango, 2018-01-25:
it's x,y,z.Mayby you confused your program's coordinate with this.

EchizenG, 2018-01-25:
@ouerkakaChango okay, another quetion: What's the direction of x,y,z? I guess it is x-right, y-up, z-forward. Is that right?

ouerkakaChango, 2018-01-25:
@EchizenG My pcl_viewer's coordinate is:x-right,y-down,z-forward。I take the KeyFrameTrajectory.txt's x,y,z as the negtive of the frame's absolute world position,so when I put them together,for each frame's point cloud,I made it's position+=(x,y,z)。

ouerkakaChango, 2018-01-26:
@EchizenG this is the function I make a frame into point cloud:
PointCloud::Ptr image2PointCloud(cv::Mat& rgb, cv::Mat& depth, CAMERA_INTRINSIC_PARAMETERS& camera)
{
	PointCloud::Ptr cloud(new PointCloud);

	for (int m = 0; m < depth.rows; m++)
		for (int n = 0; n < depth.cols; n++)
		{
			ushort d = depth.ptr<ushort>(m)[n];
			if (d == 0)
				continue;
			PointT p;

			p.z = double(d) / camera.scale;
			p.x = (n - camera.cx) * p.z / camera.fx;
			p.y = (m - camera.cy) * p.z / camera.fy;

			p.b = rgb.ptr<uchar>(m)[n * 3];
			p.g = rgb.ptr<uchar>(m)[n * 3 + 1];
			p.r = rgb.ptr<uchar>(m)[n * 3 + 2];

			cloud->points.push_back(p);
		}
	cloud->height = 1;
	cloud->width = cloud->points.size();
	cloud->is_dense = false;

	return cloud;
}