raulmur/ORB_SLAM2 #502: How mono_AR works ?

pras2020, 2018-01-04:
When monocular slam have scale ambiguity how do proper depth is estimated for placing the augmented object for AR application (i.e mono_AR in orb slam2)?

AlejandroSilvestri, 2018-01-04:
@pras2020 

As you mentioned, monocular slam doesn't know the scale.  In mono_AR the cubes are added with arbitrary scale, one related to the viewport and not to the scene.

pras2020, 2018-01-16:
If in single camera we could achieve only arbitrary scale how some papers are comparing the localization accuracy of stereo camera with monocular camera. There are few products like "dyson 360"  vacuum cleaner robot which use single camera for localization, how accurate it would be.

AlejandroSilvestri, 2018-01-16:
@pras2020 

Comparison between stereo and monocular are up to scale.  First provide a real scale to monocular map, then compute errors.

I don't know Dyson 360.  Does it need scale?  Does it need accuracy?  In general, mobile robots can use odometry info from wheels, that gives scale to the system.

Harry-675, 2018-11-14:
@AlejandroSilvestri Hello, I plan to apply ORB-SLAM2 to AR application, but the world coordinate of ORB-SLAM2 is different every time it initials, so the virtual object will not be fixed in AR application, how can I solve this problem?

AlejandroSilvestri, 2018-11-14:
@Harry-675 ,

You must learn orb-slam2's world coordinate each time.

If you can save the map (this feature is not implemented in orb-slam2, but many programmers did their own version of map serialization, and posted them in this forum), so each time you load it, it maintains the reference system.

Also, you can add an algorithm to recognize some landmark, let say a QR code, so knowing its pose you can express orb-slam2 pose to your reference system.