raulmur/ORB_SLAM2 #508: Memory leak caused by MapPoint and KeyFrame erase.

highlightz, 2018-01-12:
When erasing a MapPoint pointer or a KeyFrame pointer from the Map, the code just delete the pointer of queue, and no memory releasing operation is performed. This is a problem when the machine has not enough memory. Is there a good way to handle this?

highlightz, 2018-01-12:
Below shows the related code: 

```
void Map::EraseMapPoint(MapPoint *pMP) {
    unique_lock<mutex> lock(mMutexMap);
    mspMapPoints.erase(pMP);

    // TODO: This only erase the pointer.
    // Delete the MapPoint
  }

  void Map::EraseKeyFrame(KeyFrame *pKF) {
    unique_lock<mutex> lock(mMutexMap);
    mspKeyFrames.erase(pKF);

    // TODO: This only erase the pointer.
    // Delete the MapPoint
  }
```

xiaod17, 2018-01-17:
same question

highlightz, 2018-01-18:
In fact, I have tried to replace MapPoint* with std::shared_ptr<MapPoint>, and replace KeyFrame* with std::shared_ptr<KeyFrame>, in order to automatically manage the memory. But this makes no big difference. The destruct function of KeyFrame is never called when a KeyFrame is culled.

zc08, 2018-01-21:
I'm reading the code and trying to figure out how concurrency is handled in the system. I'm confused here, too. 
It seems the authors are aware of the resource leaking problem as indicated in the comments, but left it as is intentionally. Is there a reason to do this?

zc08, 2018-01-22:
@highlightz I think cyclic references exist. MapPoint maintains a vector that points to KeyFrames where it is visible, and KeyFrame also maintains a list of MapPoints.

highlightz, 2018-01-22:
@zc08 Yeah, you are correct. Cyclic reference does exist. I've replace std::shared_ptr<KeyFrame> with std::weak_ptr<KeyFrame>, and this technology can handle cyclic reference.

xiaod17, 2018-01-22:
@highlightz so the std::weak_ptr is useful?

highlightz, 2018-01-22:
Not really, destructor of KeyFrame is still not called. This denotes that there'is stall MapPoint associating with a Culled KeyFrame, which causes reference count of KeyFrame pointer never decreases to zero.

highlightz, 2018-01-22:
I think in the program, there is somewhere one KeyPoint in a KeyFrame(aka. 2D observation), corresponds to at lease two MapPoints.

richard-elvira, 2018-01-22:
The main KeyFrame and MapPoint is in the Map class, inside of a set container, and other references at some KeyFrame or MapPoint is pointing to its. If you want to remove the KeyFrame or MapPoint you can do it from Map, but if you do it without remove all the references in others class to the same instance you would have an access error.

highlightz, 2018-01-22:
You are right, this is why I've tried to use C++ smart pointer to manage every MapPoint* and KeyFrame*.

lixiaohang, 2018-03-17:
weak_ptr and sharded_ptr is time comsuming ,do you have any good ideal to remove the Memory leak caused by MapPoint and KeyFrame erase

arifle, 2018-10-24:
Did you guys solve this?

GiuliaMarchesi, 2020-04-15:
I am trying to solve the same problem. Did anybody solve this?

xzwang12345, 2021-01-06:
Hi @highlightz  Did you solve this? Thanks