raulmur/ORB_SLAM2 #510: process has died [pid 2262, exit code -1

pras2020, 2018-01-15:
Please help me to solve the issue 

error
"[slam-2] process has died [pid 2262, exit code -11, cmd /home/prashanthr/catkin_ws/devel/lib/orb_slam2/Stereo /home/prashanthr/catkin_ws/src/ORB_SLAM2_ROS/orb_slam2_lib/Vocabulary/ORBvoc.txt /home/prashanthr/catkin_ws/src/ORB_SLAM2_ROS/orb_slam2_ros/settings/taracali_orb.yaml true __name:=slam __log:=/home/prashanthr/.ros/log/096d2d36-f934-11e7-a8c8-c81f66027c0a/slam-2.log].
log file: /home/prashanthr/.ros/log/096d2d36-f934-11e7-a8c8-c81f66027c0a/slam-2*.log"

[image]

rcabg, 2018-01-22:
Hey there,

from your problems description it is difficult to find out the solution. 

From my experience, this kind of errors get solved introducing trace messages and trying to find the exact line when it crashes. If you find it and even that don't get why, please repost your results. 

Cheers

kanwal41g, 2020-05-01:
Hi i wanted to know if this issue got resolved. I am facing the same issue and guidance would be highly appreciated.

Charles-cyx, 2022-05-15:
> Hi i wanted to know if this issue got resolved. I am facing the same issue and guidance would be highly appreciated.

Hi, I want to know if this issue got solved? If solved, could you tell me the method. Thank you so much

mgrallos, 2023-05-30:
im facing the same problem and no other error message. how to fix this?