raulmur/ORB_SLAM2 #512: Can I use ORBMatcher.cc to do feature matching?

AldrichCabrera, 2018-01-20:
Hi. I try use orbmatcher module to do feature matching with the function the serch by initialization.  The question is if this function is good for my work or I have used map point for the performance the matcher?

AlejandroSilvestri, 2018-01-29:
Hola @AldrichCabrera ,

No se entendió la pregunta.  Quizás usaste un traductor que te la tradujo mal.  Planteala en castellano, si puedo la respondo en inglés para que los demás la entiendan.

AldrichCabrera, 2018-01-29:
Hola. Disculpen no soy muy bueno en ingles. Mi pregunta mas bien es saber como hacen el matching para una cámara monocular?

AlejandroSilvestri, 2018-01-29:
@AldrichCabrera 

There are several matching algorithms in ORB-SLAM2, each of them is slightly different.

For initialization, orb-slam2 performs an almost brute force ORB matching between initial image features descriptors and current image features descriptors.

After initialization, matches are performed between current image features and 3D mappoints descriptors.  Epipolar geometry is used to early discard pairs, avoiding slow descriptor distance computation.  To the same end, when BoW is available (for example, in keyframes) it is also used to early discard.

AldrichCabrera, 2018-01-29:
So, if I want to perform ORB matching between an initial image and current image without use MapPoint, is recommendable use search for Initialization?

AlejandroSilvestri, 2018-01-29:
@AldrichCabrera ,

Yes.  [SearchForInitialization](https://alejandrosilvestri.github.io/os1/doc/html/class_o_r_b___s_l_a_m2_1_1_o_r_bmatcher.html#aff9b6dde7878d59e334ed5ad2ddd04eb) is used with a window size 100.

AldrichCabrera, 2018-02-08:
Hi again. The SearchForInitialization give me bad match points. Could I change something in these function to obtain better points?

AlejandroSilvestri, 2018-02-08:
@AldrichCabrera 

1. Be sure your camera is calibrated.
2. Try using repeatedly ORB_SLAM2::Tracking::MonocularInitialization(), which retries initialization on each consecutive frame of a video.

AldrichCabrera, 2018-02-08:
Yes, the camera is calibrated. And I'm doing the process as you suggested, the problem is when the frames are exactly the same the max match point is 2000, but after moving the camera and having a 90% covisibility the match point drops to 400. I'd suppose the amount of matches would change proportionally to covisibility.

AlejandroSilvestri, 2018-02-08:
MonocularInitialization calls SearchForInitialization with a window size of 100.  Try with a wider window, let's say 2000.

AldrichCabrera, 2018-02-09:
It is working now. :) The covisibility give me good data. I just have to eliminate outliers to obtain best match points. Thank you very much for the support! I really appreciate your help AlejandroSilvestri.

mojovski, 2018-02-10:
Check Out the code from GSM matcher https://github.com/JiawangBian/GMS-Feature-Matcher
The code demonstrates also how to use opencv brute force matching where you can switch between detector and descriptor easily

AldrichCabrera, 2018-02-13:
@mojovski Thank you. I will keep it in mind. :+1:  :)